SolderedElectronics/Inkplate-Arduino-library #130: [Inkplate 6 Plus] touchscreen leads to horrible battery life

itobey, 2022-02-02:
I experience quite a high power draw in deep sleep when the touchscreen of my Inkplate 6 plus has been initialized beforehand. 

My tests showed that this entirely depends on the touchscreen beeing initialized (frontlight stayed disabled in all testing), even when not configuring it to wake it by touch (as displayed in this [example](https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/dev/examples/Inkplate6PLUS/Advanced_Inkplate_Features/Inkplate_Wake_up_on_touchscreen/Inkplate_Wake_up_on_touchscreen.ino)).

The goal was to have the Inkplate 6 Plus stay in deep sleep until I wake it by pressing the touchscreen, then having it send an mqtt message, refresh the screen and go back to sleep. However with a 1500 mah battery this would last only for about 3 days - and even if I don't wake it once and it stays in deep sleep all the time! If the touchscreen is not initialized the Inkplate can last in deep sleep for many weeks. So this little line here `display.tsInit(true))` compromises the entire advantage of the Inkplate.

I have attached an image of the impact of the same deep sleep code, once with the initialization of the touchscreen and once without. The power draw changes from -0,12v to roughly -0,01v (both for about 20 hours). The code only wakes it periodically to send the current voltage over mqtt. Please ignore the initial spike on each phase as this was while having the Inkplate connected and it was charging.

[image]

So my question now is this: could this behaviour be due to a bug or does this work as designed and the power draw for this feature has to be immense?

Thanks for the clarifications!

nitko12, 2022-02-08:
Can I close this now as it has been answered on the forum?
> Hello, 
> If you don't use a touchscreen then you have little current in deep sleep (something about 20uA), but if you use a touchscreen current is much higher because the touch controller in the panel has that current and we can't do anything. Because of that, we added a wake-up button on board and you don't need to use a touchscreen to wake up the microcontroller.

closed 2022-02-08 by nitko12

itobey, 2022-02-08:
Well, yeah I guess, although I'm not happy with this circumstance that my use case has been rendered impossible.