SolderedElectronics/Inkplate-Arduino-library #137: macOS upcoming release removes python 2, causes odd issue

kidGodzilla, 2022-03-01:
I use a developer preview build of macOS for work, and I think an upcoming release is going to break the compilation process.

https://developer.apple.com/documentation/macos-release-notes/macos-12_3-release-notes#Python

Basically, "python" won't be available in CLI, only python3. Not super easy to alias it and get it in your $PATH.

Further, once I was able to get python3 in my path as just `python`, I am still unable to compile a blank sketch targeting Inkplate.

I've tried nightly build of Arduino IDE and manually updating the library to match the repo, seems to be a current issue.

Kind of seems like some nuance in the Arduino IDE (Electron?) app not having access to my full $PATH. It definitely doesn't seem to have access to things I've aded to `.zshrc` (`.bash_profile`).

Feel free to tell me if I'm just missing something obvious 😅

--

### Path / which :

```
➜  ~ echo $PATH
/Users/jamesfuthey/.nvm/versions/node/v16.13.1/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/Apple/usr/bin:/Users/jamesfuthey/bin/python
➜  ~ which python
/usr/local/bin/python
```

kidGodzilla, 2022-03-01:
### Compilation of blank sketch for Inkplate 6 Plus:

```
Arduino: 1.8.20 Hourly Build 2021/12/20 07:34 (Mac OS X), Board: "Inkplate 6 plus (ESP32), Enabled, Default, 80MHz (WiFi/BT), 115200, Warn"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/jamesfuthey/Library/Arduino15/packages -hardware /Users/jamesfuthey/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/jamesfuthey/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/jamesfuthey/Documents/Arduino/libraries -fqbn=Croduino_Boards:Inkplate:Inkplate6plus:PSRAM=enabled,PartitionScheme=default,CPUFreq=80,UploadSpeed=115200,DebugLevel=warn -ide-version=10820 -build-path /var/folders/9t/_17bd0x91rb6s70jndbqdf_r0000gn/T/arduino_build_607872 -warnings=default -build-cache /var/folders/9t/_17bd0x91rb6s70jndbqdf_r0000gn/T/arduino_cache_422386 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=/Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/tools/esptool_py/3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=/Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/tools/esptool_py/3.0.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/tools/xtensa-esp32-elf-gcc/1.22.0-96-g2852398-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-96-g2852398-5.2.0.path=/Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/tools/xtensa-esp32-elf-gcc/1.22.0-96-g2852398-5.2.0 -prefs=runtime.tools.mkspiffs.path=/Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/tools/mkspiffs/0.2.3 -verbose /var/folders/9t/_17bd0x91rb6s70jndbqdf_r0000gn/T/untitled2022188680.tmp/sketch_mar01a/sketch_mar01a.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/jamesfuthey/Library/Arduino15/packages -hardware /Users/jamesfuthey/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/jamesfuthey/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/jamesfuthey/Documents/Arduino/libraries -fqbn=Croduino_Boards:Inkplate:Inkplate6plus:PSRAM=enabled,PartitionScheme=default,CPUFreq=80,UploadSpeed=115200,DebugLevel=warn -ide-version=10820 -build-path /var/folders/9t/_17bd0x91rb6s70jndbqdf_r0000gn/T/arduino_build_607872 -warnings=default -build-cache /var/folders/9t/_17bd0x91rb6s70jndbqdf_r0000gn/T/arduino_cache_422386 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=/Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/tools/esptool_py/3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=/Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/tools/esptool_py/3.0.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/tools/xtensa-esp32-elf-gcc/1.22.0-96-g2852398-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-96-g2852398-5.2.0.path=/Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/tools/xtensa-esp32-elf-gcc/1.22.0-96-g2852398-5.2.0 -prefs=runtime.tools.mkspiffs.path=/Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/tools/mkspiffs/0.2.3 -verbose /var/folders/9t/_17bd0x91rb6s70jndbqdf_r0000gn/T/untitled2022188680.tmp/sketch_mar01a/sketch_mar01a.ino
Using board 'Inkplate6plus' from platform in folder: /Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1
Using core 'Inkplate' from platform in folder: /Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1
Warning: Board breadboard:avr:atmega328bb doesn't define a 'build.board' preference. Auto-set to: AVR_ATMEGA328BB
bash -c "[ ! -f /var/folders/9t/_17bd0x91rb6s70jndbqdf_r0000gn/T/untitled2022188680.tmp/sketch_mar01a/partitions.csv ] || cp -f /var/folders/9t/_17bd0x91rb6s70jndbqdf_r0000gn/T/untitled2022188680.tmp/sketch_mar01a/partitions.csv /var/folders/9t/_17bd0x91rb6s70jndbqdf_r0000gn/T/arduino_build_607872/partitions.csv"
bash -c "[ -f /var/folders/9t/_17bd0x91rb6s70jndbqdf_r0000gn/T/arduino_build_607872/partitions.csv ] || cp /Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1/tools/partitions/default.csv /var/folders/9t/_17bd0x91rb6s70jndbqdf_r0000gn/T/arduino_build_607872/partitions.csv"
Detecting libraries used...
/Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/tools/xtensa-esp32-elf-gcc/1.22.0-96-g2852398-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -I/Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1/tools/sdk/include/config -I/Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1/tools/sdk/include/app_trace -I/Users/jamesfuthey/L [... cut ...] 1rb6s70jndbqdf_r0000gn/T/arduino_build_607872/sketch_mar01a.ino.elf
esptool.py v3.0-dev
python /Users/jamesfuthey/Library/Arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1/tools/gen_esp32part.py -q /var/folders/9t/_17bd0x91rb6s70jndbqdf_r0000gn/T/arduino_build_607872/partitions.csv /var/folders/9t/_17bd0x91rb6s70jndbqdf_r0000gn/T/arduino_build_607872/sketch_mar01a.ino.partitions.bin
exec: "python": executable file not found in $PATH
Error compiling for board Inkplate 6 plus (ESP32).
```

claud9999, 2022-07-16:
How I solved this (found via duckduckgo):
1. Install Python 3 (I use brew.sh, `brew install python`, probably needs XCode command-line tools too)
2. Create a symbolic link: `ln -s /usr/local/bin/python3 /usr/local/bin/python`
3. Start the Arduino IDE from command line: `open /Applications/Arduino.app`

robertoash, 2022-07-27:
This solution worked for me:

`sed -i -e 's/=python /=python3 /g' ~/Library/Arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1
/platform.txt`

> "The command modifies all the platform.txt files under your local installation of the ESP32 tools (~/Library/Arduino15/packages/esp32/hardware/esp32) and replaces the word python by python3 wherever it finds it"

Found it here: https://forum.arduino.cc/t/exec-python-executable-file-not-found-in-path/971847/16

KarloLeksic, 2023-04-13:
The issue is closed because there is no more activity

closed 2023-04-13 by KarloLeksic