SolderedElectronics/Inkplate-Arduino-library #143: WiFi Examples crashing

C9H13NO3-dev, 2022-03-25:
Hey everyone!

No matter what I do about this, everytime I flash a sketch using WiFi, the board seems to be crashing instantly. This is my Serial output:

> 09:30:07.385 > Rebooting...
> 09:30:07.388 > ets Jun  8 2016 00:22:57
> 09:30:07.388 >
> 09:30:07.388 > rst:0xc (SW_CPU_RESET),boot:0x32 (SPI_FAST_FLASH_BOOT)
> 09:30:07.388 > configsip: 0, SPIWP:0xee
> 09:30:07.388 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> 09:30:07.388 > mode:DIO, clock div:2
> 09:30:07.388 > load:0x3fff0018,len:4
> 09:30:07.388 > load:0x3fff001c,len:1100
> 09:30:07.388 > load:0x40078000,len:10900
> 09:30:07.388 > load:0x40080400,len:6360
> 09:30:07.388 > entry 0x400806b4
> 09:30:08.314 > Wavefrom load failed! Upload new waveform in EEPROM. Using default waveform.
> 09:30:08.485 > CORRUPT HEAP: multi_heap.c:432 detected at 0x3ffd1c98
> 09:30:08.488 > abort() was called at PC 0x4008fae7 on core 1
> 09:30:08.494 >
> 09:30:08.494 > ELF file SHA256: 0000000000000000
> 09:30:08.496 >
> 09:30:08.496 > Backtrace: 0x4008b78c:0x3ffcd840 0x4008ba09:0x3ffcd860 0x4008fae7:0x3ffcd880 0x400900ac:0x3ffcd8a0 0x40087949:0x3ffcd8c0 0x4008dc82:0x3ffcd8e0 0x400de625:0x3ffcd920 0x400dc12b:0x3ffcd950 0x400dc61c:0x3ffcd980 0x400d4ea9:0x3ffcda70 0x400d24a8:0x3ffcda90 0x400dff0a:0x3ffcdbd0 0x4008c2d2:0x3ffcdbf0

Am I doing anything wrong? What could cause this issue?

Greetings Epi

BornaBiro, 2022-03-25:
Hello C9H13NO3-dev.
Sorry to hear you have a problem with your Inkplate.
Can you give us more info like what example did you use, what Inkplate are you using, what version of the Inkplate library and Arduino Inkplate Core do you have installed, do you power your Inkplate from the battery or the USB cable?

C9H13NO3-dev, 2022-03-25:
Hey BornaBiro,
My Inkplate is an Inkplate 10, delivered yesterday. This log is after flashing the "Inkplate_WiFi_HTTP_Request" example. I downloaded the 5.7.0 inkplate library and I'm powering my Inkplate via USB-C.
Best regards, Epi

BornaBiro, 2022-03-28:
We have found the problem. Essentially, the index of waveform LUT has gone out of the boundaries of the array.
It's fixed in 93394781b203303727c72e9e3d872dcb74a8a091 and it should work fine now.

If you have any other problems or questions, feel free to ask.

C9H13NO3-dev, 2022-03-28:
Thank you, this actually solved my problem with the examples crashing!

closed 2022-03-28 by C9H13NO3-dev