SolderedElectronics/Inkplate-Arduino-library #145: Version 5.7.0 has breaking changes from 5.5.1

lanrat, 2022-03-25:
There is a breaking change between version 5.5.1 and 5.7.0 that I've bene unable to identify.

I use my [HomePlate](https://github.com/lanrat/homeplate) library to display screenshots over the network on my Inkplate10.

It's been working perfectly for months on version 5.5.1. I recently tested version 5.7.0, and it [no longer works](https://github.com/lanrat/homeplate/issues/3). I've reviewed the [diff](https://github.com/e-radionicacom/Inkplate-Arduino-library/compare/5.5.1...5.7.0) between the versions and don't see anything too obvious that might have changed.

I have a hunch it has something to do with the MCP registers, but I have yet to find it.

Are there any changes between these versions that could be related?

Other than looking though the commit history, is there a more detailed change log between releases or release notes?

lanrat, 2022-03-27:
I've done more debugging and narrowed it down to [this commit](https://github.com/e-radionicacom/Inkplate-Arduino-library/commit/269f287a42ab53b9cffe1f9eed8165b6188086b0) which results in a corrupt heap when using most methods of the WiFi class.

```
15:14:34.025 > CORRUPT HEAP: multi_heap.c:432 detected at 0x3ffb607c
15:14:34.031 > abort() was called at PC 0x400914a3 on core 1
15:14:34.036 > 
15:14:34.036 > ELF file SHA256: 0000000000000000
15:14:34.039 > 
15:14:34.039 > Backtrace: 0x4008d100:0x3ffb1c20 0x4008d379:0x3ffb1c40 0x400914a3:0x3ffb1c60 0x40091b74:0x3ffb1c80 0x400827b9:0x3ffb1ca0 0x400827ea:0x3ffb1cc0 0x40089ca1:0x3ffb1ce0 0x4000beaf:0x3ffb1d00 0x4008dc46:0x3ffb1d20 0x40121da6:0x3ffb1d40 0x401228a5:0x3ffb1d60 0x40144841:0x3ffb1d80 0x400d2371:0x3ffb1dc0 0x400d28c5:0x3ffb1df0 0x400d1941:0x3ffb1f00 0x400d1587:0x3ffb1f70 0x400db8be:0x3ffb1fb0 0x4008e37e:0x3ffb1fd0
15:14:34.425 >   #0  0x4008d100:0x3ffb1c20 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
15:14:34.425 >   #1  0x4008d379:0x3ffb1c40 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
15:14:34.425 >   #2  0x400914a3:0x3ffb1c60 in multi_heap_assert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380
15:14:34.425 >       (inlined by) multi_heap_malloc_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:432
15:14:34.425 >   #3  0x40091b74:0x3ffb1c80 in multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:305
15:14:34.425 >   #4  0x400827b9:0x3ffb1ca0 in heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:361
15:14:34.425 >   #5  0x400827ea:0x3ffb1cc0 in heap_caps_malloc_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:361
15:14:34.425 >   #6  0x40089ca1:0x3ffb1ce0 in _malloc_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:37
15:14:34.425 >   #7  0x4000beaf:0x3ffb1d00 in ?? ??:0
15:14:34.425 >   #8  0x4008dc46:0x3ffb1d20 in xQueueGenericCreate at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:2038
15:14:34.425 >   #9  0x40121da6:0x3ffb1d40 in sys_mbox_new at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:217
15:14:34.425 >   #10 0x401228a5:0x3ffb1d60 in tcpip_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:501
15:14:34.425 >   #11 0x40144841:0x3ffb1d80 in tcpip_adapter_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1255
15:14:34.425 >   #12 0x400d2371:0x3ffb1dc0 in tcpipInit() at /Users/lanrat/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:2114
15:14:34.425 >   #13 0x400d28c5:0x3ffb1df0 in WiFiGenericClass::mode(wifi_mode_t) at /Users/lanrat/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:2114
15:14:34.425 >       (inlined by) WiFiGenericClass::mode(wifi_mode_t) at /Users/lanrat/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:537
15:14:34.425 >   #14 0x400d1941:0x3ffb1f00 in keepWiFiAlive(void*) at src/network.cpp:59
15:14:34.425 >   #15 0x400d1587:0x3ffb1f70 in setup() at src/main.cpp:108
15:14:34.425 >   #16 0x400db8be:0x3ffb1fb0 in loopTask(void*) at /Users/lanrat/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:18
15:14:34.425 >   #17 0x4008e37e:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
15:14:34.425 > 
``` 

My new best guess is some incomparability between the WiFi class and the new EEPROM library being included.

BornaBiro, 2022-03-28:
Hello, lanrat!

I believe it's the same problem as this one #143 so try to use the newest version from the master branch and it should work fine.

lanrat, 2022-04-04:
Thanks.

Pulling the latest from master, or specifically 93394781b203303727c72e9e3d872dcb74a8a091, did resolve the issue.

closed 2022-04-04 by lanrat