SolderedElectronics/Inkplate-Arduino-library #148: Strange "dots" when using 1Bit instead of 3Bit mode

maltejahn, 2022-04-25:
Hi,

i own a inplate 6. I download a generated png file from a local server. Generaly this works without issues in 3Bit mode, but in 1Bit mode I get dots everywhere on the screen.

I would like to use 1Bit mode because sometimes I get no Wifi . I would like to keep the screen with the "old" information, but add a "NoWifi" symbol. This,  I think, should be possible in partial mode which is only available in 1Bit mode

I use Platformio

Btw, i tried to post this in the Forum, but I cant get the Accout to work an if I try to contact an Admin i receive this error: "Sorry but all email related functions have been disabled."

Here is the reduced code which leads to the described error

```
#include "Inkplate.h"
#include "time.h"
#include <WiFi.h>

WiFiClient espClient;
Inkplate display(INKPLATE_1BIT);

#define uS_TO_S_FACTOR 1000000 // Conversion factor for micro seconds to seconds
#define TIME_TO_SLEEP_SHORT  120      // How long ESP32 will be in deep sleep (in seconds)

// User settings
const char ssid[] = "ssid"; // 2.4GHz, no Mesh
const char *password = "password";     // Your WiFi password
const char *hostname="display_hallway";

// URL using http://ip:port/filename.png
const char *image_url = "http://192.168.0.243:81/inkplate_6_hallway.png";

void setup()
{

  Serial.begin(115200);
  Serial.println("Display starts");

  // set sleep timer to short sleep, if download was successful long sleep time will be overwritten later
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP_SHORT * uS_TO_S_FACTOR);
  Serial.println("Initial Sleep Time set to " + String(TIME_TO_SLEEP_SHORT) + " Seconds");

  display.begin();        // Init Inkplate library (you should call this function ONLY ONCE)

  // Join wifi
  Serial.println("joining wifi");
  WiFi.setHostname(hostname);
  if(display.joinAP(ssid, password)==true)
    {
      Serial.println("joined wifi");
   
    }
  else
    {
      Serial.println("ERROR: problems joining wifi");
    }
    
  if(display.drawImage(image_url, display.PNG, 0, 0)==true)
    {
      Serial.println("got imagefile from url");

    }
  else
    {
      Serial.println("ERROR: getting imagefile from url");
      display.setCursor(300, 300); display.print("Error getting imagefile from url");

    }

  display.display(); // Send everything to display (refresh the screen)
    

  Serial.println("Going to sleep now");
  delay(1000);
  Serial.flush();    
  // go to deep sleep
  esp_deep_sleep_start();
  Serial.println("ERROR: This will never be printed");

 }

void loop()
{
    // Never here, as deepsleep restarts esp32
}
```

Here is the original file, is this file wrong?
[image]

[image]

BornaBiro, 2022-05-09:
Hi maltejahn!

The reason for these strange dots is that the dithering is enabled by default on 1bit mode as well as in 3bit mode while using drawImage function. If you want to disable it, you can set the fifth argument in the [drawImage](https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/master/src/include/Image.h#L67) function to false. This should disable it.
`bool drawImage(const char *path, const Format &format, const int x, const int y, const bool dither = 1, const bool invert = 0);`

Hope this fixes your problem.

maltejahn, 2022-05-18:
Hi,

thanks for your reply and sorry for answering that late.

The dots are gone (thanks), but.... 

1Bit mode + 
display.drawImage(image_url, display.PNG, 0, 0, false, false)
now the image seems to be inverted. Would be ok (but dont understand why) if inverting would work.

I tried some of your bmp files (neowise) with 1Bit Mode. There, the file is loaded correctly when inverting colors
display.drawImage(image_url, display.BMP, 0, 0, false, true)
So, for BMP files inverting works, but not for png

Is there a problem/error with inverting PNGs?

The problem for me is, that i cant edit the "script" that build the image(PNG). So i´m not able to just switch to another file format. I also tried to modify the color space.
But I also cant there where the "invert" is used in the drawPngFromWeb function

Thanks

BornaBiro, 2022-05-31:
Unfortunately, you're right. For some reason, it will not flip the colors on the PNG file, but it works fine on BMP images.
We will try to fix the issue.

DavidSichau, 2022-11-26:
I run into the same problem when drawing 1Bit png images. Is there a fix in progress?

KarloLeksic, 2023-04-15:
Hi,

Apologies for the late reply.

In the past few months, there have been a lot of upgrades and improvements in the library. Could you please try the latest version of the library and board definition?

rsoric, 2024-03-25:
Closing this as there's no more activity

closed 2024-03-25 by rsoric