SolderedElectronics/Inkplate-Arduino-library #149: display.drawImage Fails on Images Generated from Google Slides

leoherzog, 2022-05-13:
I have an Inkplate 10 project that's using the sample [`Image_Frame`](https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/master/examples/Inkplate10/Projects/Image_Frame/Image_Frame.ino) code, but to pull a thumbnail image from Google Slides instead of Unsplash. Attached is an example of the generated image:

[image]

Example URL that I'm dynamically fetching via Apps Script and using in `drawImage()` (not sure how long this will remain valid):

[`https://lh5.googleusercontent.com/hdLLrW3Lo5Z6qs8NEZfgf8GvnWElxnPadUUVbWKm3Qs9e7RNqTS2xR98vIFaLqUE8qPeXEzzP-r9C3MRsZpoQIJ5nO5-ndkDlvGBZypsL3fRsqtK_TTifinKQwVkT3zj0uKcP2c6ahqVTO2gd7YsWZoZIy_5i60QOkuR_cg7O5nuemI9puELDlET9TUElSCxZMRLMNfpYCLjyNzp02RJstshrgrnKc1qSk4iolO_jjIURspTPSo=s1200`](https://lh5.googleusercontent.com/hdLLrW3Lo5Z6qs8NEZfgf8GvnWElxnPadUUVbWKm3Qs9e7RNqTS2xR98vIFaLqUE8qPeXEzzP-r9C3MRsZpoQIJ5nO5-ndkDlvGBZypsL3fRsqtK_TTifinKQwVkT3zj0uKcP2c6ahqVTO2gd7YsWZoZIy_5i60QOkuR_cg7O5nuemI9puELDlET9TUElSCxZMRLMNfpYCLjyNzp02RJstshrgrnKc1qSk4iolO_jjIURspTPSo=s1200)

When I call `Serial.println(display.drawImage(url, display.PNG, 0, 0));`, a `0` is printed, indicating that there was an error. How do I determine what the error is? Are there certain types of PNG files that aren't supported? The `url` variable is correct when logged.

leoherzog, 2022-05-13:
I tried taking this `test.png` file and hosting it elsewhere, then hard-coding _that_ URL into the `drawImage()` function. Still no dice.

leoherzog, 2022-05-17:
After trying various ways of exporting the file, building a proxy with Cloudflare Workers, and trying Github Pages hosting, I learned that the function works correctly when pulling the image from a simple HTTP server on my local machine. Sure enough, any host with `https` does not work, but `http` works fine. Turned off SSL enforcement on my Github Pages hosting, hard-coded `http` in my URL, and it's working fine.

Now, to figure out why SSL isn't working...

BornaBiro, 2022-05-31:
Hello [leoherzog]!

I'm glad that you found the problem.

The only thing that comes to mind is that drawImage() uses a basic HttpClient for getting the data from the Internet (in this case image) doesn't have any support for HTTPS and SSL. Maybe using WiFiClientSecure in [downloadFile() function](https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/master/src/include/NetworkClient.cpp#L86) would fix the problem, but that means rewriting whole downloadFile() function or making a new one only for HTTPS and SSL.

But the weird thing is and tried to display a few random images from the Internet that have https:// in the link on the Inkplate and they work perfectly fine.

leoherzog, 2022-09-15:
Revisiting this. Here's code that doesn't work:

```c
// Next 3 lines are a precaution, you can ignore those, and the example would also work without them
#ifndef ARDUINO_INKPLATE10
#error "Wrong board selection for this example, please select Inkplate 10 in the boards menu."
#endif

#include "Inkplate.h"

Inkplate display(INKPLATE_3BIT);

void setup() {
  
    Serial.begin(115200);
    display.begin();

    Serial.println("Battery: " + String(display.readBattery()));

    while (!display.joinAP("myssid", "mypassword")) {
      Serial.println("Connecting to WiFi...");
    }
    Serial.println("Connected!");

    display.clearDisplay();
    Serial.println(display.drawImage("https://placehold.jp/1200x825.jpg", display.JPG, 0, 0));
    
    display.display();

    Serial.println("Going to sleep...");
    delay(100);
    esp_sleep_enable_timer_wakeup(30 * 60 * 1000 * 1000); // 30 minutes
    esp_deep_sleep_start();

}

void loop() {
    // Never here, as deepsleep restarts esp32
}
```

I have also tried proxying through Cloudflare Workers in order to disable SSL, turn gzip on and off, stripping all headers, etc. I can't for the life of me get this to work :(

My Cloudflare Workers URL that also doesn't work:
`http://inkplate.herzog.tech/image.jpg`

The serial monitor:

```shell
rst:0x1 (POWERON_RESET),boot:0x32 (Sets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x32 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10900
load:0x40080400,len:6360
entry 0x400806b4
Wavefrom load failed! Upload new waveform in EEPROM. Using default waveform.
Battery: 3.26
Connecting to wifi.
Connected!
0
Going to sleep...
```

I _really_ wish the `display.drawImage()` function returned a more helpful error than `0`.

leoherzog, 2022-09-15:
This URL doesn't work when I hard-code it:

`https://loremflickr.com/1200/825`

But if I proxy it through Cloudflare Workers, remove all headers, and disable SSL, it works?

`https://inkplate.herzog.tech/image.jpg`

leoherzog, 2022-09-15:
Any ideas, @nitko12 or @davidzovko?

leoherzog, 2022-09-15:
or @BornaBiro or @gogi640?

davidzovko, 2022-09-16:
@ifadiga Can you take a look, please?

BornaBiro, 2022-11-07:
@leoherzog The reason why it doesn't work with https://loremflickr.com/1200/825 and it works on http://inkplate.herzog.tech/image.jpg it's because of HTTPS usage. For now, the Inkplate library doesn't support HTTPS, but it will in the near future.

For some deeper investigation, we need to see what is going on under the hood of the ESP32 HTTP library.

KarloLeksic, 2023-04-13:
The issue is closed because there is no more activity

leoherzog, 2023-04-13:
This is still an issue.

rsoric, 2023-05-09:
Hi @leoherzog, we've looked into this a bit more. 

On the /dev branch of the Inkplate Library repo you can now find a function 'setFollowRedirects'. From the .ino file It's called as such:

`display.setFollowRedirects(HTTPC_FORCE_FOLLOW_REDIRECTS);`

There are several options you can set, as you can see in ESP32's Arduino library:
`HTTPC_DISABLE_FOLLOW_REDIRECTS` , `HTTPC_STRICT_FOLLOW_REDIRECTS` and `HTTPC_FORCE_FOLLOW_REDIRECTS` .

After implementing this function and calling it to 'FORCE_FOLLOW_REDIRECTS', your link worked for me with the draw image function. Note that you should call the version of the function which specifies the image type, just as you have in your example:

`display.drawImage("https://loremflickr.com/1200/825", display.JPG, 0, 0)`

I also managed to display "https://placehold.jp/1200x825.jpg" without problems.

Let us know if it works, again, make sure to use the version of the library on /dev.

rsoric, 2023-05-09:
Hi @misel228 

I tried the URL you provided and that one seems to be running into issues even with the newly implemented function. After forcing Inkplate to use HTTP1.0 for the GET request, it seems to download and show the first chunk of the image (horizontally the top 20% of the image).

The primary focus of the draw image function is to show a specific image pointed to by a URL, file path or memory location. Unfortunately, we can't do much with changing how files are downloaded because it's related to ESP32's core library which handles it. If you have a specific suggestion for a method to try and resolve this issue, let us know, we'll test it for you.

misel228, 2023-05-09:
I was able to solve my problem by simply refactoring my server side code so that the content is not sent chunked. I would assume the others commenters here did the same so the URLs work now and don't show the affected problem.

Also, I understand that actually fixing it can be a lot of trouble. The content-length header is probably used to allocate the memory before the actual download. Not knowing how much you receive makes the code much more complex.

What you could do however - and should in my opinion - is put a big red warning flag in the documentation: "Chunked transfers are not supported" or something similar.

rsoric, 2023-05-10:
Good to hear it's working for you now. 

We will add it in the docs, as more information is always valuable for users.

closed 2023-05-10 by rsoric