SolderedElectronics/Inkplate-Arduino-library #150: BT Connection to slave device fail

dedesite, 2022-06-08:
Hello,
I'm opening a separated bug for this problem already mentioned in this issue #98.

I'm trying to pair my Inkplate 6 with a BT 3.0 Keyboard and it doesn't work, even with the "ESP32 Wroover Module" board from this definition : https://dl.espressif.com/dl/package_esp32_index.json

Here is the code I use, you may want to change the address or the name to get it working :
```cpp
#include "Inkplate.h"
#include "Arduino.h"
#include "Fonts/Open_Sans_SemiBold_32.h"
#include "BluetoothSerial.h"

const GFXfont *text2_font = &Open_Sans_SemiBold_32;

BluetoothSerial SerialBT; //Object for Bluetooth

String MACadd = "DC:2C:26:E4:67:D6";
uint8_t address[6]  = {0xDC, 0x2C, 0x26, 0xE4, 0x67, 0xD6};
String name = "Bluetooth USB keyboard";

Inkplate display(INKPLATE_1BIT);

int incoming;
bool connected;

void setup()
{
    display.begin();        // Init library (you should call this function ONLY ONCE)
    display.clearDisplay(); // Clear any data that may have been in (software) frame buffer.
    //(NOTE! This does not clean image on screen, it only clears it in the frame buffer inside ESP32).
    display.display(); // Clear everything that has previously been on a screen
    display.setCursor(150, 320);
    display.setTextSize(1);
    display.setFont(text2_font);

    // Tentative BT
    Serial.begin(9600); //Start Serial monitor in 9600 also tried with other rates
    SerialBT.begin("ESP32_KEYBOARD", true); //Name of your Bluetooth Signal
    display.print("Bluetooth Device is Ready to Pair");
    display.display();
    // connect(address) is fast (upto 10 secs max), connect(name) is slow (upto 30 secs max) as it needs
    // to resolve name to address first, but it allows to connect to different devices with the same name.
    // Set CoreDebugLevel to Info to view devices bluetooth address and device names
    //connected = SerialBT.connect(address);
    connected = SerialBT.connect(name);

    if(connected) {
      display.println("Connected Succesfully!");
      display.partialUpdate(false, true);
    } else {
      while(!SerialBT.connected(10000)) {
        display.print("Failed to connect.");
        display.partialUpdate(false, true);
      }
    }
}

void loop()
{
 
}
```

Is there any progress on your side to get BT working in both slave and master mode with your board definition ?
Thanks,
Andréas

dedesite, 2023-01-30:
Any update on this issue ?

KarloLeksic, 2023-04-13:
We tried to recreate the example with newer board definitions, but unfortunately, it didn't work properly. If there is progress, we will make an example in the Inkplate library.

dedesite, 2023-04-14:
Thank you for the update, I hope this will work someday.

rsoric, 2024-03-25:
Closing this issue as there's no more activity

closed 2024-03-25 by rsoric