SolderedElectronics/Inkplate-Arduino-library #157: no value changing of 'Inkplate::readBattery()'

TedMGN, 2022-08-28:
Hi,
I am very happy with my Inkplates 6.
I also use a battery to be able to use the device without a cable.
To check the status of the battery I use the function 'Inkplate::readBattery()'.
But the function always returns the same value without restart.
It means
- if the Inkplate runs without reset the value does not change.
- if I use the on/off switch oder the reset button. After the restart, the value is different from before.
Do you know that problem? How I can fix this wrong reading?
Or do I have the possibility (in example a function) to make a hard or soft restart of the Inkplate?
Thanks for your effort.
Best regards
Detlef

KarloLeksic, 2023-04-15:
Hi,

Sorry for the late reply, we were quite busy. In the future, we will try to answer quickly.

Can you please provide a code example? We have never had a similar situation. Inkplate reads the voltage at the time the function is called.

What values do you get? Is it a normal voltage or some random values?

TedMGN, 2023-04-16:
Hi,
thanks for your response. I tried to make a code example for the bug, but the code example works fine.
Now I just checked my code again and after a while I found a bug inside my code base.
Sorry for the wrong issue.
Thanks for your efforts.
Best regards
Detlef

closed 2023-04-16 by TedMGN