SolderedElectronics/Inkplate-Arduino-library #158: Inkplate 6 COLOR Boards update not yet released?

TheWebMachine, 2022-09-29:
It appears the latest Boards version is still 1.0.1 in Arduino, which does not contain the 6 COLOR definitions. I just got my 6 COLOR yesterday and can't program it because the board is missing. When will this be released?

Arduino 1.8.19 on Ubuntu
Current Inkplate Lib v5.7.1
Current Inkplate Boards v1.0.1 (latest showing in Boards Manager)

misel228, 2022-09-30:
Same here on Arduino 2.0.0

When I use a text search for "color" on the boards definition file I cannot find anything at all

https://raw.githubusercontent.com/e-radionicacom/Croduino-Board-Definitions-for-Arduino-IDE/master/package_Croduino_Boards_index.json

davidzovko, 2022-10-01:
Hi, we did not push the 6COLOR software. Sorry about that. The whole team is currently traveling and on a Maker Faire. When we are back in the office, first thing next week, we will publish this. Sorry for the delay.

szehnder, 2022-10-04:
Any update on this?  I'd love to play w/ the InkPlate Color but can't do much w/out the board showing up in arduino.

TheWebMachine, 2022-10-04:
It is only Tuesday. David is pretty good at keeping things going. I'd rather things be right than quick. 😉

misel228, 2022-10-04:
I also contacted them through the support form from crowd supply and have already gotten a response.

They fixed the documentation at https://inkplate.readthedocs.io/en/latest/get-started.html

The new board definition URL für Arduino is here: https://github.com/e-radionicacom/Dasduino-Board-Definitions-for-Arduino-IDE

I have not had the time to test it yet, though. But maybe it's of help to you.

TheWebMachine, 2022-10-04:
I'll test it after dinner. 😎

davidzovko, 2022-10-05:
What @misel228 said is 100% correct :) we forgot to replace the board definition link [here](https://inkplate.readthedocs.io/en/latest/get-started.html). The correct link is [this one](https://github.com/e-radionicacom/Dasduino-Board-Definitions-for-Arduino-IDE). 

Now everything should be okay. Anything else appears, let us know!

closed 2022-10-05 by davidzovko

AnthonyDiGirolamo, 2022-11-17:
The README in this repo says

> [Install Inkplate board definition](https://github.com/e-radionicacom/Croduino-Board-Definitions-for-Arduino-IDE/blob/master/README.md) - add Inkplate 6/10/6PLUS as a board into your Arduino IDE. Follow the instructions on the [link](https://e-radionica.com/en/blog/add-inkplate-6-to-arduino-ide/).

https://e-radionica.com/en/blog/add-inkplate-6-to-arduino-ide prompts to install the old board definition too: https://raw.githubusercontent.com/e-radionicacom/Croduino-Board-Definitions-for-Arduino-IDE/master/package_Croduino_Boards_index.json instead.

Please update the readme here with the correct instructions. Thanks!

BornaBiro, 2022-11-30:
@AnthonyDiGirolamo these are old board definitions, You must use Soldered Dasduino Board Definitions that You can find [here](https://github.com/SolderedElectronics/Dasduino-Board-Definitions-for-Arduino-IDE).

README.md on Croduino Board Definitions page has been updated and it prompts users to use new Soldered Board Definitions.

> NOTE: Croduino Board Definitions are not updated anymore. Please use new Soldered Dasduino Board Definitions for Dasduino and Inkplate boards. You can find them [here](https://github.com/SolderedElectronics/Dasduino-Board-Definitions-for-Arduino-IDE).