SolderedElectronics/Inkplate-Arduino-library #161: Not drawing BMP images from URL correctly.

chrisjtwomey, 2022-10-08:
I have this piece of code:

```
display.begin();
display.clearDisplay();
display.display();

if (!display.drawImage(url, 0, 0, false, false)) {
    display.println("Image open error");
}
display.display();
```

When `url = "http://192.168.1.11:/homepage.png"`, Inkplate10 display draws image as I'd expect:

[image]

When `url = "http://192.168.1.11:/homepage.bmp"`, the image drawn is almost completely black:

[image]

I thought maybe there was some funny business happening with how Inkplate is interpreting the background but inverting the image doesn't seem to make much of a difference:

[image]

I've attached both the png and bmp images I've been testing with below if anyone else wants to try them:
[homepage.zip](https://github.com/e-radionicacom/Inkplate-Arduino-library/files/9739028/homepage.zip)

What's interesting is that trying the same files with a Waveshare board using GXEPD2, I get behaviour similar to PNG here. In the end, I've gotten around this by just using PNG, but still interested in the cause of this.

KarloLeksic, 2023-04-15:
Hi,

Try updating the Inkplate library to the latest and also the board definition. If that doesn't solve the problem, maybe the problem is in the color palette. Convert the image to 24-bit bmp as that seems to work best.

rspurlock, 2023-05-06:
I believe the pull request titled "Add missing default arguments and fix 1-bit BMP draw image from buffer" will fix this BMP display issue as that was the problem I was having trying to draw monochrome (1-bit) BMP images.

rsoric, 2024-03-25:
Closing this as there's no more activity, please reopen the issue if you think it needs more attention

closed 2024-03-25 by rsoric