SolderedElectronics/Inkplate-Arduino-library #164: Error handling revamp suggestion

misel228, 2022-10-23:
Hi everyone,

while toying around with my Inkpad 6 COLOR I noticed that the error handling in the library is quite rudimentary.

The functions return 1 on success and 0 on failure. This information is not really helpful, though. I'm actually more used to have functions return 0 on success and >0 on error. This way each error that each function could return can be a unique number, that together with an "errno" style function could even return a user readable error message.

I understand that this would require a massive overhaul of the entire library but given it's fragility I'm certain it would help a lot.

But maybe I'm also missing something else entirely. Is there a reason you have chosen this path? Or is it possible to do a step through with a debugger to find the errors?

All best,
Misel

BornaBiro, 2022-11-07:
Hi misel228.

The only way it would be useful and also possible without overcomplicating the whole library would be the part for the image decoding and grabbing images from the web (although JPG and PNG decoders are libraries, and if the library itself doesn't use any error handling, we can't do much about that). Everything else cannot be easily handled, like refreshing the panel (and all its functions), initialization of the panel (on IP6Color), drawing graphics etc.

Besides that many users would not use some kind of error handling (for most users, classic succ/fail is ok in our experience).

No, unfortunately, it's not possible to use the debugger, because all the I/O lines of the ESP32 are used up by e-paper panel (and mixing e-paper and debugger signals is not a good idea).

Can You explain to me what part of the code You have a problem with that you need step-by-step debugging?

misel228, 2022-11-07:
I've tried loading a few images through HTTP and HTTPS without success. The examples show up the [image frame example](https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/master/examples/Inkplate6COLOR/Projects/Image_Frame/Image_Frame.ino) works just fine but as soon as I point the image URL to something on my server it fails.

And it's really frustrating that it doesn't even tell me where. Is it the HTTP or HTTPS path, is it the image format? I don't know because the lib doesn't tell me.

BornaBiro, 2022-11-07:
As mentioned on https://github.com/SolderedElectronics/Inkplate-Arduino-library/issues/149 HTTPS does not work on the current version of the Inkplate library and Inkplate Arduino Core. We will try to add this feature later.

We will implement some kind of error handling on that part, at least on the HTTP part, but this will take some time.
Can You send the problematic images, so that we can investigate this issue?

misel228, 2022-11-07:
Going back to my test project - after 2 weeks - I realized that my HTTP connection was automatically upgraded to HTTPS because I set it up years ago and forgot.

There's still an issue when loading JPGs but for that I will open a separate issue.

Even though it now works - at least partially - a proper error message would have immediately pointed me in the right direction.

I thought about backwards compatibility and maybe something like a global variable with an error code in it is better than my initial proposal. This way the methods could still return 1 on success and on 0 for failure the inclined user can get said error code and look it up somewhere.

KarloLeksic, 2023-04-15:
Hi,

as I answered you on one of the previous issues, the same here. In the past time, a lot of work was done on the improvements of the Inkplate library, so this may have been solved. Apologies once again for the inactivity. We were busy, but we will be more active in the future.

rsoric, 2024-03-25:
Closing this as there's no more activity, please reopen the issue if you think it needs more attention

closed 2024-03-25 by rsoric