SolderedElectronics/Inkplate-Arduino-library #165: [Inkplate 6 Plus]

claria, 2022-11-01:
Hi,

i try to use partial update (basic example code), where i update parts of the screen and keep the other parts of the screen. 

While the latest written text has a bright color, after an partial update at another place, the color of the existing text becomes much lighter. This is most pronounced for the first partial update, further partial update calls only have a much smaller effect.

The picture shows text drawn via partial update while at each iteration, the y offset is increased. So the lowest text is the latest written. 

I was expecting that this happens only after multiple partial updates and not with the first one already. Is this expected and is there something i can do to reduce the effect?

[image]

Btw, i really like your product!

BornaBiro, 2022-11-07:
Hello claria!

I am very glad to hear that You are enjoying using our product!

Unfortunately, this is the limitation of the e-paper itself and our Inkplate library. Also, on some e-paper panels, it fades much more than on others.

The partial updates on Inkplate work in such a way that there is a comparison between what is on the screen and what needs to be written on it and only sends that difference to the screen (Black pixels to white pixels, or white pixels to black pixels). Because of that, it losses some charge on the parts of the screen that does not need to be changed.

You can try to minimize this effect by keeping the e-paper power supply on (but this will dramatically increase power consumption) by using the function partialUpdate(0, 1); You can keep the e-paper power supply on when you need to do partial updates very frequently, but turn it off when you don't need them (by calling einkOff() or parialUpdate() without parameters).

Hope this helps.

claria, 2022-11-12:
Hi @BornaBiro,

using einkOff() / einkOn() did not have a big effect. 

However i found a "-2.50" on a sticker on the flat ribbon cable of the epaper display and just tried to set this as vcom via the Factory_Programming_VCOM sketch. This had a very large effect and i can now do up to 6 partial refreshs with only a small drop of quality.

Is the number on the flat ribbon cable the vcom voltage and is it okay to set this VCOM voltage?

matthewmiddlebrook, 2023-01-20:
@claria I think you may have fixed my biggest gripe with the Inkplate 6 Plus!

Previously a compromise I found was to flash the display white to fix the image rather than doing a partial or full refresh. I've been hoping that this partial update problem was a software issue that would be fixed at some point since I've never heard of E Ink displays having this problem. I took a risk to change the vcom based on your comment and it seems to work beautifully, and even better, the change persists even after wiping the code so I can use ESPHome with this fix. Now I can display a clock and weather data without having the display flash all the time! I'm not sure of the long term effects of this change so I'll report back after a while if there's issues that arise.

BornaBiro, 2023-03-13:
Sorry for the late reply, somehow I missed updates on this issue.

I'm glad that the issue has been resolved. It seems that the VCOM has not been properly programmed in TPS65186 eink PMIC and it is set to -1.27V on the VCOM or even worse on 0V. Tried VCOM Factory Programming example on a few Inkplates and it was programmed successfully, so I really don't know what happen on your Inkplates.

```
 I've been hoping that this partial update problem was a software issue that would be fixed 
 at some point since I've never heard of E Ink displays having this problem.
```
Eink epapers do not have this problem but bear in mind that we are not using proper eInk waveforms for driving panels since they are proprietary to the eInk Corporation. Even if we could get our hands on the waveforms, we could not drive the panel that fast (since eInk waveform specs. require 85Hz refresh rate, and we could only get maybe 15Hz - 20Hz), and also everything is reverse-engineered, so all these leads to the image quality not being as same as on some ebook reader.

Writing VCOM voltage written on the sticker on the ribbon cable should not create any problems.

I'm closing this issue since it's been resolved. If you encounter more problems with the partial updates, feel free to reopen the issue.

closed 2023-03-13 by BornaBiro