SolderedElectronics/Inkplate-Arduino-library #166: Compile Error: 'class WiFiClientSecure' has no member named 'setInsecure'

AlexW00, 2022-11-02:
When trying to compile [this example project](url) for to setup [arduinoWebSockets](https://github.com/Links2004/arduinoWebSockets) on my Inkplate 10, I get the following compilation error:
```
/Users/awDocuments/GitHub/inkplate-printer/firmware/libraries/arduinoWebSockets-2.3.6/src/WebSocketsClient.cpp:234:30: error: 'class WiFiClientSecure' has no member named 'setInsecure'
                 _client.ssl->setInsecure();
                              ^
```

Compling the code for a regular [esp32](https://github.com/espressif/arduino-esp32) board works. Compiling for Inkplate does not.

Info:
- arduinoWebSockets: 2.3.6 (installed via zip from GitHub releases)
- inkplate boards: 1.0.1
- inkplate library: 5.7.1
- board: inkplate 10

BornaBiro, 2022-11-07:
Hi AlexW00.

This is due to the library arduinoWebSockets and esp32 Arduino core update. For now, if You can, roll down to the older version of the arduinoWebSockets library, or wait a few days until when a new version of the Inkplate Arduino Core will be released that should support these functions (it will be based on ESP32 Arduino Core v2.0.5).

KarloLeksic, 2023-04-13:
The issue is closed because there is no more activity

closed 2023-04-13 by KarloLeksic

rsoric, 2023-05-19:
Hi @AlexW00 

This should work fine, check [this ](https://github.com/SolderedElectronics/Inkplate-Arduino-library/issues/203#issuecomment-1554045084)reply I gave to @petitchou who brought this issue again to our attention recently

-Rob