SolderedElectronics/Inkplate-Arduino-library #167: JPEG doesn't load

misel228, 2022-11-07:
I've been trying to create an image rotator from my personal library and I can't get my Inkplate 6COLOR to load jpeg files.

The code below is a minimal version that tries to load a jpeg off of my server. What you cannot see is that the URL is actually hiding a PHP script with mod_rewrite to serve a JPG file or a PNG file depending on the chosen file extension.

If I change the end of the URL to .png and use the drawImage line with the format `display.PNG` everything works and I see three red letters on my display. But the code below shows only a blank screen when it should show three blue letters "JPG".

I'm very well aware that this code might be completely correct and that my server code needs enhancement. I have already corrected the mime type from "image/jpg" to "image/jpeg" and I have disabled an automated protocol elevation to HTTPS.

I don't know why the screen stays blank. Please advise.

```

  #include "Inkplate.h"
  
  Inkplate display;
  
  const char ssid[] = "ABCDEF"; // Your WiFi SSID
  const char *password = "xxxxx";     // Your WiFi password
  const char *url = "http://home.misel.ws/image_frame.jpeg"; // .jpg doesn't work either
  
  void setup()
  {
      Serial.begin(115200);
      display.begin();
  
      // Join wifi
      display.joinAP(ssid, password);
      Serial.println("joined wifi");
  
      bool result = display.drawImage(url, display.JPG, 0, 0, true, false);
      //    bool result = display.drawImage(url, display.PNG, 0, 0, true, false);
  
      Serial.println(result);
  
  
      display.display();
  
      Serial.println("Going to sleep");
      esp_sleep_enable_timer_wakeup(15ll * 60 * 1000 * 1000);
      esp_deep_sleep_start();
  
  
  }
  
  void loop()
  {}
```

KarloLeksic, 2023-04-15:
Hi,

We apologize for the late reply. Is this still a problem? There have been a few updates to the Inkplate library since the day you reported the problem. Try to download the latest version of the Inkplate library and the latest version of the board definition and then run the code. If it is still a problem, we will try to solve it as soon as possible.

misel228, 2023-04-15:
Thanks for getting back

Somewhere deep in another issue someone mentioned that chunked transfers were not supported. I've since modified the PHP script to return the file size in the header which in turn caused Apache to not send it chunked and then it worked.

trm42, 2023-04-27:
@misel228 thanks for the pointer. Same happened for me as well. 

For the posterity the fix with plain PHP is to call

```php
header('Content-Length: ' . strlen($imageContent));
```

Ofc if one is using some nice framework like Laravel or Symfony, then the same should be done through its response objects and so forth.

rsoric, 2024-03-25:
Closing this as there's no more activity, please reopen the issue if you think it needs more attention

closed 2024-03-25 by rsoric