SolderedElectronics/Inkplate-Arduino-library #169: readBattery returns 0.00 for inkplate 6Color

jakobwesthoff, 2022-11-19:
I have two different inkplate 6Color boards here, which, if driven by battery always respond with `0.00` to a `readBattery()` call. The position of the call itself does not change anything (before `display.begin()` or after).

I first thought it might be a hardware problem, but as I tested it with two different 6Color boards, I am thinking maybe there is something wrong with the library. I have a lot of different other Inkplates, and I don't have a problem with those. (Multiple Inkplate 6 and Inkplate 10s).

This is a minimal example to reproduce the issue:
```
#include "Inkplate.h"

Inkplate display;

void setup()
{
  Serial.begin(115200);

  sleep(5);

  log_d("before begin: %f", display.readBattery());
  display.begin();
  log_d("after begin: %f", display.readBattery());
}

void loop()
{
  sleep(3600);
}
```

Output: 
```
[  5476][D][main.cpp:15] setup(): before begin: 0.000000
[  5655][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[  5973][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  6006][D][main.cpp:17] setup(): after begin: 0.000000
```

misel228, 2022-11-20:
Same here: I also extended the float precision and it's really all zeroes. Something's going wrong.

Battery level: 0.0000000000

jakobwesthoff, 2022-11-20:
@misel228 good to know I am not the only one experiencing this. I would assume the ADC connection is switched via mosfets, and then run through a voltage devider, to not consume power when disabled. Maybe the circuit is connected differently on the 6color and is not properly activated, before reading the adc.

however I am just guessing here. Hopefully somebody from soldered can bring light into this, and hopefully it is fixable somehow. It would be a problem for any kind of safe battery usage, if the battery level can’t be checked properly.

misel228, 2022-11-20:
I was googling around because and am even more confused than before. One of the example mentions to recalibrate the ADC of the ESP32. But I couldn't find out how. Additionally, there seem to be many issues regarding the reliability of the ESP32's ADC.

Additionally, I tried to compare the sources of the 6COLOR PCBs with the PCB sources of the other boards. They're supposed to be open source, right? But ... where are the sources? I couldn't find them.

jakobwesthoff, 2022-11-21:
@misel228 The calibration is usually done either using y LUT, which takes quite some memory or a polynominal function. Both of those scenarios are used to correct the fact, that the reference voltage of the ADC within the ESP32 is not always exactly the same. Furthermore the development of the measurements is not strictly linear. Using either of the techniques the measurements can be made more accurate, as well as with multisampling and other solutions, to remove noise from the measurement.

However all of this is definitely not the problem here, as the ADC simply gets no reading at all. The results are looking like the adc line is pulled low instead of being connected to the battery. That's why I guessed there might be something wrong with the activation of the circuit in order to take a measurement.

It would be really great, if the @BornaBiro, @nitko12, @gogi640 or any other person from solderedcom could take a look at this, as I am quite confident, that we are not making any progress here without more detailed knowledge of the circuits on this board.

BornaBiro, 2022-11-30:
Hello everyone. I'm sorry for the late reply.

Currently, we are having some problems with the new IO expander IC. Due to the chip shortage, we needed to change the IO expander from MCP23017 to the new PCAL6416A. We will release a new version of the library (and also the Arduino Inkplate board def.) to fix this problem very soon.

For now, if you need to read the battery voltage, you can use readBatteryVoltage() function written below:

```
double readBatteryVoltage()
{
  uint16_t rawADC;
  double voltage;
  
  // Set PCAL P1-1 to output. Do a ready-modify-write operation.
  pcal6416ModifyReg(0x07, 1, 0);

  // Set pin P1-1 to the high -> enable MOSFET for battrey voltage measurement.
  pcal6416ModifyReg(0x03, 1, 1);

  // Wait a little bit
  delay(5);

  // Read analog voltage. Battery measurement is connected to the GPIO35 on the ESP32.
  rawADC = analogRead(35);

  // Turn off the MOSFET.
  pcal6416ModifyReg(0x03, 1, 0);

  // Calculate the voltage
  voltage = rawADC / 4095.0 * 3.3 * 2;

  // Return voltage.
  return voltage;
}

void pcal6416ModifyReg(uint8_t _reg, uint8_t _bit, uint8_t _state)
{
  uint8_t reg;
  uint8_t mask;
  const uint8_t pcalAddress = 0b00100000;
  
  Wire.beginTransmission(pcalAddress);
  Wire.write(_reg);
  Wire.endTransmission();

  Wire.requestFrom(pcalAddress, 1);
  reg = Wire.read();

  mask = 1 << _bit;
  reg = ((reg & ~mask) | (_state << _bit));
  
  Wire.beginTransmission(pcalAddress);
  Wire.write(_reg);
  Wire.write(reg);
  Wire.endTransmission();
}
```

Just call readBatteryVoltage() in the code. pcal6416ModifyReg() is a helper function.

Hope this helps.

CounterPillow, 2022-11-30:
@BornaBiro Thank you! This code works. As a small improvement:
```
Wire.requestFrom(pcalAddress, (uint8_t) 1);
```
gets rid of the compiler warning.

Looks like the MCP23017 chips will only be back in stock in Spring, so switching was probably the right choice since NXP's supply appears to be more stable.

As an aside for anyone running into it, the missing code for the new I/O expander also means the touch button examples won't work.

jakobwesthoff, 2022-12-01:
Thanks. I can confirm this works now as expected. Combined with PR #171 it yields even better results 👍

```cpp
#ifdef ARDUINO_INKPLATECOLOR
// Currently needed as the new port multiplexer is not supported by the default library yet:
// https://github.com/SolderedElectronics/Inkplate-Arduino-library/issues/169#issuecomment-1331716568
// Combined with the following open PR: https://github.com/SolderedElectronics/Inkplate-Arduino-library/pull/171/commits/124462fdf49963d7227881cc4a28c28f4ff40f6e
void pcal6416ModifyReg(uint8_t _reg, uint8_t _bit, uint8_t _state)
{
  uint8_t reg;
  uint8_t mask;
  const uint8_t pcalAddress = 0b00100000;

  Wire.beginTransmission(pcalAddress);
  Wire.write(_reg);
  Wire.endTransmission();

  Wire.requestFrom(pcalAddress, (uint8_t)1);
  reg = Wire.read();

  mask = 1 << _bit;
  reg = ((reg & ~mask) | (_state << _bit));

  Wire.beginTransmission(pcalAddress);
  Wire.write(_reg);
  Wire.write(reg);
  Wire.endTransmission();
}

double readBatteryVoltage()
{
  // Set PCAL P1-1 to output. Do a ready-modify-write operation.
  pcal6416ModifyReg(0x07, 1, 0);

  // Set pin P1-1 to the high -> enable MOSFET for battrey voltage measurement.
  pcal6416ModifyReg(0x03, 1, 1);

  // Wait a little bit
  delay(5);

  // Read analog voltage. Battery measurement is connected to the GPIO35 on the ESP32.
  uint32_t batt_mv = analogReadMilliVolts(35);

  // Turn off the MOSFET.
  pcal6416ModifyReg(0x03, 1, 0);

  // Calculate the voltage
  return (double(batt_mv) / 1000 * 2);
}
#endif
```

BornaBiro, 2023-01-16:
@jakobwesthoff
Yes, you're right! Using analogReadMilliVolts() will get you far more precise results, thank you for mentioning that.

@CounterPillow unfortunately yes. But you can access them in a similar way.
Here is the code

```C++

const uint8_t pcalAddress = 0b00100000;

uint8_t getTouchpadState(int8_t _t)
{
    // Variables for storing PCAL register content
    uint8_t _pcalInputReg;

    // Only 0, 1 and 2 are vaild arguments.
    if (_t > 4 || _t < 1) return -1;
  
    // Set touchpad I/O to input. Do a read-modify-write operation.
    // NOTE: Touchpad P1-3 is connected to the touchpad 1.
    pcal6416ModifyReg(0x07, 3 + _t - 1, 1);

    // Enable the pull-up resistor on the touchpad I/O pin.
    pcal6416ModifyReg(0x47, 3 + _t - 1, 1);

    // Set the pull-up resistor on touchpad I/O pin.
    pcal6416ModifyReg(0x49, 3 + _t - 1, 1);

    // Read the state of the pin.
    pcal6416GetReg(0x01, &_pcalInputReg, 1);

    // Return the pin state.
    return ((_pcalInputReg >> (3 + _t - 1)) & 1);
}

void pcal6416ModifyReg(uint8_t _reg, uint8_t _bit, uint8_t _state)
{
  uint8_t reg;
  uint8_t mask;

  Wire.beginTransmission(pcalAddress);
  Wire.write(_reg);
  Wire.endTransmission();

  Wire.requestFrom(pcalAddress, (uint8_t)1);
  reg = Wire.read();

  mask = 1 << _bit;
  reg = ((reg & ~mask) | (_state << _bit));

  Wire.beginTransmission(pcalAddress);
  Wire.write(_reg);
  Wire.write(reg);
  Wire.endTransmission();
}

void pcal6416GetReg(uint8_t _reg, uint8_t *_r, uint8_t _n)
{
    // Set the internal PCAL pointer to the address of the input register.
    Wire.beginTransmission(pcalAddress);
    Wire.write(_reg);
    Wire.endTransmission();

    // Read the register
    Wire.requestFrom(pcalAddress, _n);

    // Wait until all bytes are received
    while (Wire.available() != _n);

    // Save the bytes
    Wire.readBytes((uint8_t*)_r, _n);
}

```

Calling the function getTouchpadState() will return the state of the selected touchpad, so calling getTouchpadState(1) will read the state of the touchpad 1.

We will try to push a new version of the library and Arduino Inkplate Board Def. with support for the PCAL I/O expander as soon as possible. If you can't wait for that, it's already on Github so you can use it, but note that everything is still not 100% functional.
Arduino Library with PCAL support https://github.com/SolderedElectronics/Inkplate-Arduino-library/tree/PCAL-Support
Arduino Inkplate Borad Def. for PCAL Support: https://github.com/SolderedElectronics/Dasduino-Board-Definitions-for-Arduino-IDE/tree/dev

BornaBiro, 2023-03-07:
I'm closing this issue since there is a new version of the Inkplate library that should solve this problem.

If someone is still having an issue with reading the battery or reading the touchpad, or anything else related to the I/O of the Inkplate6Color, feel free to reopen the issue.

closed 2023-03-07 by BornaBiro