SolderedElectronics/Inkplate-Arduino-library #173: WiFi.status() returns WL_NO_MODULE (255)

AlexW00, 2023-01-13:
When running [this example](https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/master/examples/Inkplate10/Advanced_Inkplate_Features/Inkplate_WiFi_HTTP_Request/Inkplate_WiFi_HTTP_Request.ino), the Inkplate (10) device crashes after calling `WiFi.scanNetworks();`.
Printing `WiFi.status() ` before the the `scanNetwork` call returns 255, which stands for WL_NO_MODULE.
Is this a hardware problem? Are there any ways to fix it?

BornaBiro, 2023-01-16:
Hello @AlexW00!

Can You give me more info about Your setup (OS Type, Inkplate Library Version (and branch), Arduino Inkplate Borad Def.)?
Also, a printout from the serial would be useful.

Only HW issue can be a problem with the ESP32 module itself, but this is rarely an issue.

BornaBiro, 2023-01-26:
@AlexW00 Any updates?

AlexW00, 2023-01-26:
Sorry for the late reply The issue has been resolved after using a different USB cable to upload the code. Don't ask me why, but it works now.

closed 2023-01-26 by AlexW00

BornaBiro, 2023-01-30:
No problem, I'm glad You solved the issue. 🙂 

Bear in mind that Inkplate boards can create high current spikes while refreshing the screen and connecting to the WiFi (especially Inkplate 10 with its 9.7" display), so using a good quality cable with low voltage drop is a must.