SolderedElectronics/Inkplate-Arduino-library #174: Could I upgrade esp32 module to an esp32s3

zhangtemplar, 2023-01-16:
Just curious, could  I upgrade esp32 module to an esp32s3? So I could have:
1. more power, ram and storage;
2. USB client connection, e.g., as usb c portal monitior?

Bests!

BornaBiro, 2023-01-17:
Hello @zhangtemplar 

First, they are not pin-to-pin compatible, so some kind of adapter must be made to be able to connect ESP32-S3 to the Inkplate board. And ESP32-S3 module with PSRAM must be used.

Maybe, but we didn't test it. We did some tests on ESP32-S2 and we got the same results (in some cases even worse, I think it was Inkplate 10 Code, Grayscale Full Refresh, but it was a long time ago, so do not quote me on that). Speed is all that matters - the faster MCU, the better. If you have a slower main clock, getting proper grayscale is almost impossible (even 1 bit partial update can be hard to get). As far as I can see ESP32-S3 has 240MHz for the main clock and it's dual-core, so same as ESP32 WROVER, so in theory, could be possible.

For USB Host/Device, bear in mind that ESP32 on Inkplate has almost all of GPIOs used up just to drive the e-paper. If the ESP32-S3 has the same number of GPIO, you will have to remove some functionality from the Inkplate (like microSD card) and move e-paper lines to the different GPIOs to free USB lines. Again, could be possible, but a lot of work.

For future questions like this, please contact us at https://soldered.com/contact/

zhangtemplar, 2023-01-18:
I see. Thanks for your answer @BornaBiro

closed 2023-01-18 by zhangtemplar