SolderedElectronics/Inkplate-Arduino-library #176: Inkplate 6 COLOR Hardware Files

zachstence, 2023-02-12:
I am looking for the hardware files (circuit schematics, 3D printed case) for the Inkplate 6 COLOR, but don't see them anywhere in this GitHub organization. Will they be released any time soon?

[image]

I know this probably isn't the best place for this issue, but I wasn't sure where else it could go since the repo I'm looking for doesn't exist.

ifadiga, 2023-02-12:
Hi,
We are going to redo all boards, incl. Inkplate 6PLUS, in KiCAD and Soldered branding, and release them here on our GitHub.

We finished all of Inkplate 10 and published the files. Also, when we finish Inkpalte 6COLOR, we will post CAD files for it here on GitHub.

On the [link ](https://github.com/SolderedElectronics/Inkplate-6-hardware/issues/5), you can download only schematics for Inkplate boards.

KarloLeksic, 2023-04-13:
The issue is closed because there is no more activity. All files will be posted when it will be finished.

closed 2023-04-13 by KarloLeksic