SolderedElectronics/Inkplate-Arduino-library #177: GIF Image Support

leoherzog, 2023-02-12:
Currently, features like [`display.drawImage()`](https://inkplate.readthedocs.io/en/latest/arduino.html#inkplate-drawimage) support PNG, JPEG, and BMP. Please add support for still GIF images.

BornaBiro, 2023-02-14:
Hello @leoherzog.

Thank You for the advice, we will definitely consider this feature for future upgrades in the Inkplate Library, but at the moment there will be no GIF support.

closed 2023-02-27 by BornaBiro

leoherzog, 2023-02-27:
This is... completed?

BornaBiro, 2023-02-28:
This issue is closed due to our policy; if there is no activity or any kind of response on the specific issue for more than two weeks, it will be closed (the only exception to this rule are the issues that are breaking the functionality of the Inkplate library and we are currently working on the fix).

And do not worry, your suggestion is on our Inkplate library improvements wishlist and we will consider adding that feature, but at the moment we are not able to do that due to other things that need to be added or fixed in the Inkplate library.

In the future, if You have any other suggestions for improvements to the Inkplate library, please feel free to send Your suggestions to the https://soldered.com/contact/.