SolderedElectronics/Inkplate-Arduino-library #178: High Deep Sleep current draw for InkPlate10

K-J-Q, 2023-02-15:
Measured the current consumption of InkPlate10 using PPK2, and got 8mA in deep sleep.... for wake on touchpad example

ifadiga, 2023-02-16:
Can you send a picture to us to know which Inkplate10 you have?
We have old Inkplate10 (e-radionica) and Soldered Inkplate 10, and they have small differences in the hardware design. Because of that, we must know what you have that we can help you to find what is in the problem.

K-J-Q, 2023-02-17:
Ivan,

Thanks for getting back to me. Please view the attached picture

Best Regards,
Jian Quan

ifadiga, 2023-02-17:
Hi, 
You didn't attach a picture. Can you put it here on GitHub?

K-J-Q, 2023-02-17:
[image]

ifadiga, 2023-02-17:
I measured the current on my device, and it is 23uA in sleep. How do you measure current? Did you use a battery or USB port?

If you measure with USB then the current will be higher because the USB is connected to a USB-UART converter, which needs 7-8mA when working.

You can connect 5V to a VIN pin if you have a power supply, or you can use a battery, then the board must have a current of about 23uA.

K-J-Q, 2023-02-17:
I have measured with the battery power as an input.

What program did you run ?

ifadiga, 2023-02-20:
Hi,
I used the same example but added a line to isolate GPIO_12.

Can you measure current with [this](https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/master/examples/Inkplate10/Advanced_Inkplate_Features/Inkplate_Low_Power/Inkplate_Low_Power.ino)example?

K-J-Q, 2023-02-23:
Got the desired output! thanks for your help. (connected to 3V3 pad instead of VIN)

[image]

closed 2023-02-23 by K-J-Q