SolderedElectronics/Inkplate-Arduino-library #184: Inkplate10_Wake_Up_On_Touchpads not working

DHueter, 2023-03-02:
Hello everyone,

I try to use the Inkplate 10 in deep sleep and wake up the Inkplate when a touchpad is pressed to change the content on the display. Therefore I tried the example [Inkplate10_Wake_Up_On_Touchpads](https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/master/examples/Inkplate10/Advanced/DeepSleep/Inkplate10_Wake_Up_On_Touchpads/Inkplate10_Wake_Up_On_Touchpads.ino) first, unfortunately without success.
As in previous issues the wake up button works without any problems. The touchpads themselves work with the Inkplate10_Read_Touchpads example.

My Board is defined as ARDUINO_INKPLATE10 and not as ARDUINO_INKPLATE10V2 and it looks the same as in [Issue 178](https://github.com/SolderedElectronics/Inkplate-Arduino-library/issues/178).

Since the defined TOUCHPAD_WAKE_MASK (int64_t(1) << GPIO_NUM_34) isn't used, I assume that there are indeed mistakes in this example.
It doesn't work either if I change line 62 from (1ULL << 36) to (1ULL << 34).

I hope someone can help me with this. It's probably not that hard to solve the problem if you know what you're doing, and it seems I don't. ;)
Kind regards and thanks for any help

BornaBiro, 2023-03-06:
Hello @DHueter!

It seems that there is indeed a mistake with GPIO indexing since GPIO_36 is used on the wake button and its active low, but for wake-up on touchpads it should be
`esp_sleep_enable_ext1_wakeup(TOUCHPAD_WAKE_MASK, ESP_EXT1_WAKEUP_ANY_HIGH);`
since INTB from the MCP23017 I/O expander is connected to the GPIO_34 of the ESP32 and it's active high.

We will fix this as soon as possible. Thank You for pointing that out.

DHueter, 2023-03-06:
Hi @BornaBiro,

thanks for your response, but I think it's not that easy.
If I replace this line, the board keeps restarting forever without touching anything. The wakeup reason is `ESP_SLEEP_WAKEUP_EXT1`, since it's `esp_sleep_enable_ext1_wakeup(...)` I can imagine that is correct.

I already invested a lot of hours last week and tried out a lot of things, including things where I don't know what I'm doing anymore. ;) I hope someone else finds the solution.

Kind regards

BornaBiro, 2023-03-07:
I run this example on my Inkplate 10 board (old version, blue PCB with MCP23017 as I/O Expander) and it works as expected.
```
/*
   Inkplate10_Wake_Up_On_Touchpads example for Soldered Inkplate 10
   For this example you will need USB cable and an Inkplate 10
   Select "e-radionica Inkplate10" or "Soldered Inkplate10" from Tools -> Board menu.
   Don't have "e-radionica Inkplate10" or "Soldered Inkplate10" option? Follow our tutorial and add it:
   https://soldered.com/learn/add-inkplate-6-board-definition-to-arduino-ide/

   Here is shown how to use I/O Expander and ESP interrupts to wake up the MCU from deepsleep when touchpad is pressed.

   Want to learn more about Inkplate? Visit www.inkplate.io
   Looking to get support? Write on our forums: https://forum.soldered.com/
   11 February 2021 by Soldered
*/

// Next 3 lines are a precaution, you can ignore those, and the example would also work without them
#if !defined(ARDUINO_INKPLATE10) && !defined(ARDUINO_INKPLATE10V2)
#error "Wrong board selection for this example, please select e-radionica Inkplate10 or Soldered Inkplate10 in the boards menu."
#endif

#include <Inkplate.h>

// Conversion factor for micro seconds to seconds
#define uS_TO_S_FACTOR 1000000
// Time ESP32 will go to sleep (in seconds)
#define TIME_TO_SLEEP 30

// bitmask for GPIO_34 which is connected to I/O Expander INT pin
#define TOUCHPAD_WAKE_MASK (int64_t(1) << GPIO_NUM_34)

// Initiate Inkplate object
Inkplate display(INKPLATE_1BIT);

// Store int in rtc data, to remain persistent during deep sleep
RTC_DATA_ATTR int bootCount = 0;

void setup()
{
    Serial.begin(115200);
    display.begin();

// Different I/O expander has been used on Soldered Inkplate V2 board and original Soldered Inkplate.
#ifdef ARDUINO_INKPLATE10
    display.setIntOutput(1, false, false, HIGH, IO_INT_ADDR);
    display.setIntPin(PAD1, RISING, IO_INT_ADDR);
    display.setIntPin(PAD2, RISING, IO_INT_ADDR);
    display.setIntPin(PAD3, RISING, IO_INT_ADDR);
#elif ARDUINO_INKPLATE10V2
    display.setIntPin(PAD1, IO_INT_ADDR);
    display.setIntPin(PAD2, IO_INT_ADDR);
    display.setIntPin(PAD3, IO_INT_ADDR);
#endif

    ++bootCount;

    // Our function declared below
    displayInfo();

    // Go to sleep for TIME_TO_SLEEP seconds
    esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);

    // Enable wakup from deep sleep on gpio 36
    esp_sleep_enable_ext1_wakeup(TOUCHPAD_WAKE_MASK, ESP_EXT1_WAKEUP_ANY_HIGH);

    // Go to sleep
    esp_deep_sleep_start();
}

void loop()
{
    // Never here
}

// Function that will write number of boots and boot reason to screen
void displayInfo()
{
    // First, lets delete everything from frame buffer
    display.clearDisplay();

    // Set text cursor and size
    display.setCursor(10, 280);
    display.setTextSize(2);

    display.print(F("Boot count: "));
    display.println(bootCount, DEC); // Print the number

    // Set next line cursor position
    display.setCursor(10, 320);

    // Display wake up reason
    esp_sleep_wakeup_cause_t wakeup_reason;
    wakeup_reason = esp_sleep_get_wakeup_cause();
    switch (wakeup_reason)
    {
    case ESP_SLEEP_WAKEUP_EXT0:
        display.println("Wakeup caused by external signal using RTC_IO");
        break;
    case ESP_SLEEP_WAKEUP_EXT1:
        display.println("Wakeup caused by external signal using RTC_CNTL");
        break;
    case ESP_SLEEP_WAKEUP_TIMER:
        display.println("Wakeup caused by timer");
        break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD:
        display.println("Wakeup caused by touchpad");
        break;
    case ESP_SLEEP_WAKEUP_ULP:
        display.println("Wakeup caused by ULP program");
        break;
    default:
        display.println("Wakeup was not caused by deep sleep");
        break;
    }

    display.display();
}
```

For me, it seems like the touchpads are the problem. Sometimes they can give a false reading, meaning that the touchpad will detect touch even when there is none. It can happen if the touchpads are pressed while powering on the Inkplate board since the touchpad IC is being calibrated in the first few milliseconds after power-up.

You can check a few things:
- Check the jumpers for the touchpads, all should be soldered where there is a white mark below two pads (or in other words, middle and right jumper pads must be soldered)
- Check the voltage between each touchpad jumper and GND. If no touchpads are pressed voltmeter should show 0V.
- Check IntB jumper
- Check the voltage on IntB jumper. While no touchpads are pressed, should be 0V. When the touchpad is pressed, it should momentarily jump to 3.3V
- Check the version of the board definition for the Inkplate and the version of the library if it's up-to-date. The Inkplate library should be v6.0.0 and the Inkplate board definition should be v3.0.1

closed 2023-04-13 by KarloLeksic

alexdmejias, 2023-06-23:
I'm having the same issue as OP. Looking at the suggestions posted by @BornaBiro I'm not sure what `IntB` is, can anyone enlighten me?