SolderedElectronics/Inkplate-Arduino-library #188: Inkplate 10 not charging

werdnum, 2023-03-11:
Hi folks,

I have one of the Inkplate 10s from the original crowd funding round, I guess I received it late 2021.

I'm finding that the battery does not seem to charge when the unit is plugged into the wall (I'm mostly testing using a USB C-C cable attached either to my laptop or to a PD charger. The device will run, but the battery voltage does not increase, and one time I emptied the battery completely and then plugged the device in overnight, and the battery was still empty afterwards.

The battery in question seems to be chargeable by an external device (DFRobot DFR0446) but not by the Inkplate itself. I've tried various variations of having the Inkplate off or on.

Do you have any particular troubleshooting steps to try to figure out why this isn't working?

misel228, 2023-03-11:
If I understand it correctly, you've tested only PD aware power supplies.

I'm not sure about the Inkplate 10, but I know for a fact that my Inkplate 6 COLOR doesn't have the required resistors for these smart power supplies to detect a 5V sink. In this case the PD standard requires the power supply to shut off and so it doesn't charge.

I have to charge my Inkplate with a dumb charger and a USB-A to USB-C cable.

Soldered Electronics is in good company with this fault they're certainly not the first ones to do that and I fear they won't be the last. This is especially sad since this topic has already entered the mainstream a few years ago when the first iteration of the Raspberry Pi 4 (the first with a USB-C port) had the same error.

Here is just one article about the issue with a few references https://www.tomshardware.com/news/raspberry-pi-4-usb-c-update

werdnum, 2023-03-12:
> If I understand it correctly, you've tested only PD aware power supplies.
>
> I'm not sure about the Inkplate 10, but I know for a fact that my Inkplate 6 COLOR doesn't have the required resistors for [...]

Hmm, thanks for this. Let me try an A-to-C cable on a USB-A port and get back to you.

werdnum, 2023-03-13:
I tried a regular A-to-C cable. I found that the Inkplate will run off it, but it only seems to charge up to 3.14V or thereabouts before stopping.

See attached screenshot of the voltage over time according to Home Assistant.

![Screenshot of battery voltage over time as measured by ESPHome and stored in Home Assistant. The graph shows the voltage fairly stable over time at around 3.6V until it drops precipitously to 3V and disappears. A little later, the graph rises rapidly from 3V up to around 3.1V, but it seems to top out a little above 3.1V](https://user-images.githubusercontent.com/271070/224609093-26dce280-010d-4e4e-a6dd-fba3710c4348.png)

ifadiga, 2023-03-13:
Hi, 
This seems like your Inkplate has a faulty charger. Can you connect Inkpalte to USB and measure the voltage on battery pins without a connected battery?  The voltage needs to be between 4.1 and 4.2V.

BornaBiro, 2023-03-13:
@misel228 There are separate resistors on the CC1 and CC2 on the USB-C connector as seen on the [schematic](https://github.com/SolderedElectronics/Inkplate-10-hardware/blob/main/Schematics%2C%20Gerber%2C%20BOM/v1.0/Inkplate10%20schematics%20v1.0.pdf).
[image]
I tried multiple Inkplate 10 with multiple USB-C to USC-C cables and it powers up and changes the battery just fine.

@werdnum
Does it run of the battery? And what kind of battery did you use? There are some batteries that have reversed polarity, so by connecting this battery to the JST connector, you could easily damage the charger IC. If this is not the case, then is definitely a faulty MCP73831T charger IC.

werdnum, 2023-03-13:
> Does it run of the battery? And what kind of battery did you use? There are some batteries that have reversed polarity, so by connecting this battery to the JST connector, you could easily damage the charger IC. If this is not the case, then is definitely a faulty MCP73831T charger IC.

Yes, it does run off the battery. It just doesn't seem to charge.

It's this kind of battery: https://core-electronics.com.au/polymer-lithium-ion-battery-1000mah-38458.html. I haven't used any other kind of battery.

> This seems like your Inkplate has a faulty charger. Can you connect Inkpalte to USB and measure the voltage on battery pins without a connected battery? The voltage needs to be between 4.1 and 4.2V.

Let me pull it apart and give this a go and get back to you...

werdnum, 2023-03-13:
Got the case off and a multimeter out.

- multimeter agrees that battery voltage is 3.11V.
- When disconnecting the battery, the voltage across the VBatt terminals starts at 2.6V and drops down to zero continuously as the multimeter is connected, over the course of a minute or so - so perhaps a capacitor is being drained?

Attached a couple of quick photos in case something jumps out at you.

[image]
[image]
[image]
[image]

ifadiga, 2023-03-13:
Hi, 
I see in the picture that you don't have a charger IC on the board.

Please contact us at contact@soldered.com, send photos, and write that you don't have a charger IC on your Inkplate board and you will get a replacement board.

Stanko, 2023-04-05:
Sorry to bump this issue, but I have the same problem.

To be fair I connected the battery before I swapped it's polarity. When I swapped it, Inkplate works fine on battery, but the charge LED doesn't come on when it is plugged in.

If I managed to damage something, is it possible to fix it.

Here are two photos in a case they help:
|Battery power|Plugged in|
|-|-|
|[image]|[image]|

It can be hard to see, but on both images blue power LED is on.

Thank you!

ifadiga, 2023-05-10:
Hi,
If you have a problem with charging you probably damaged the charger IC. That is a little IC near the CHRG LED. You can change it but be careful because if you heat PCB you can damage the e-paper panel. We use MCP73831T for the charger.

Stanko, 2023-05-10:
Thank you! I'll try that.