SolderedElectronics/Inkplate-Arduino-library #189: Inkplate 10 bypass power button

guillempages, 2023-03-25:
I have an Inkplate 10, that I plan to embed in an enclosure without access to the power button.
I've found out, that when the device has been unpowered for a while (talking about few seconds/minutes here), whenever it gets power over the USB port again, I still need to press the power button to wake it up/boot it.

Is there a way of bypasisng this behaviour, so that it can automatically boot when power is supplied via USB?

I'm thinking about bridging the contacts of the power switch, to create an "always-pressed" scenario, but I'm not sure if this would break/burn something on the long term.

Is there a better option to achieve my goal of "auto-power-on"?

guillempages, 2023-03-25:
In case it is relevant, I have the "old" model, with the blue PCB and the touch pads

KarloLeksic, 2023-03-28:
Hi @guillempages,

All Inkplates have pads that you can short with a 0-ohm resistor or simply a piece of wire and Inkplate will always be powered.
In the image below are marked pads on older Inkplate 10.

[image]

closed 2023-03-28 by KarloLeksic

guillempages, 2023-03-30:
Thanks a lot @KarloLeksic! That's exactly what I was looking for. I was not able find that information or read it from the design files. This should solve my problem.