SolderedElectronics/Inkplate-Arduino-library #193: Bad display module?

fridgecow, 2023-04-07:
I bought an Inkplate 10 (with battery and case) from Soldered but I've been struggling to get it to display anything. The first thing I did when I got the board was flash one of the Deep Sleep examples, but when I couldn't get it to work, I flashed the Inkplate 10 Peripheral Mode example. I'm using the board definitions from: https://github.com/SolderedElectronics/Dasduino-Board-Definitions-for-Arduino-IDE .

I can interact with the board, e.g I have no problem receiving `OK`, temperature, display mode, battery voltage. Eventually I tried turning the panel on and off with the `#Q(0)*` and `#Q(1)*` commands, which resulted in this:

[image]

However, clear commands (`#K(1)`), draw commands, etc, all do nothing. The more I turn the panel on and off, the worse the bands get.

It did look fine when it arrived:

[image]

However I have not done anything except remove the case to check the hardware.

I had to pay quite a lot of import duty so would love to get this fixed :)

fridgecow, 2023-04-07:
I finally got something displaying (the first examples weren't working I guess because of https://github.com/SolderedElectronics/Inkplate-Arduino-library/issues/191 , then peripheral mode wasn't working because I never sent `#L(1)*` to flush the buffer.

However, the stripes have remained, so I still think the display module is bad:
[image]

The lines persisted during & after the Burn-In fix example.

fridgecow, 2023-04-07:
As it happens, the lines don't appear immediately on flushing to the display, but creep in over time (maye 30s-1m). I tried reseating the flex ribbon but it did not help.

fridgecow, 2023-04-07:
The lines appear in both 1BIT and 3BIT mode.

KarloLeksic, 2023-04-11:
Hi @fridgecow,

We are sorry for the problem.
We suspect that the problem is in the Vcom voltage. Can you please run an Inkplate10_Factory_Programming_VCOM but before, comment on the lines that I also commented:

`    //if (isFirstStartup)
    //{
        // Test all the peripherals
        //testPeripheral();

        do
        {
            // Get VCOM voltage from serial from user
            uint8_t flag = getVCOMFromSerial(&vcomVoltage);

            // Show the user the entered VCOM voltage
            Serial.print("Entered VCOM: ");
            Serial.println(vcomVoltage);
            display.print(vcomVoltage);
            display.partialUpdate();

            if (vcomVoltage < -5.0 || vcomVoltage > 0.0)
            {
                Serial.println("VCOM out of range!");
                display.print(" VCOM out of range!");
                display.partialUpdate();
            }

        } while (vcomVoltage < -5.0 || vcomVoltage > 0.0);

        // Write VCOM to EEPROM
        display.pinModeInternal(IO_INT_ADDR, display.ioRegsInt, 6, INPUT_PULLUP);
        writeVCOMToEEPROM(vcomVoltage);
        EEPROM.write(EEPROMaddress, 170);
        EEPROM.commit();

        display.selectDisplayMode(INKPLATE_3BIT);

        // Select the waveform via serial
        // Write "OK" to confirm input
        int result = 1;
        //do
        //{
        //    result = getWaveformFromSerial(&selectedWaveform);
        //} while (result != 0);

        // Write waveform to EEPROM
        //waveformEEPROM.waveformId = INKPLATE10_WAVEFORM1 + selectedWaveform;
        //memcpy(&waveformEEPROM.waveform, waveformList[selectedWaveform], sizeof(waveformEEPROM.waveform));
        //waveformEEPROM.checksum = display.calculateChecksum(waveformEEPROM);
        //display.burnWaveformToEEPROM(waveformEEPROM);
    //}
    //else
    //{
    //    display.einkOn();
    //    vcomVoltage = (double)(readReg(0x03) | ((uint16_t)(readReg(0x04 & 1) << 8))) / -100;
    //    display.getWaveformFromEEPROM(&waveformEEPROM) ? waveformEEPROM.waveformId : -1;
    //}`

This is the sketch that we program to our Inkplates when we make it. It tests it, writes VCOM voltage to the TPS PMIC, and also writes waveform to EEPROM memory. With these comments, we skip testing. It only writes VCOM voltage to the TPS.

Open the Serial Monitor and enter **-1.8**. If the VCOM is programmed successfully, you will see a message on the Serial Monitor and you will get the initial screen as when you bought it.

If that was the problem, the lines will no longer be shown across the screen.

fridgecow, 2023-04-15:
Thank you for the help. The panel actually improved slightly after leaving
24 hours, and then healed completely after running the VCOM programming as
suggested.

KarloLeksic, 2023-04-25:
The issue is closed because the problem is solved and there is no more activity