SolderedElectronics/Inkplate-Arduino-library #195: Inkplate 10 SD Card Error

mattncsu, 2023-05-01:
Hello,
Is there a special kind of microSD card required for Inkplate 10? I have tried a [32gb SDXC](https://www.amazon.com/dp/B09WB35BXS) and a [2gb SD card](https://www.amazon.com/dp/B001BNNZXO) in FAT and FAT32.  I get "SD Card Error!" on the screen when I try the _Inkplate10_SD_Pictures_ and _Inkplate10_SD_TXT_Read_ examples.

Thanks!

rsoric, 2023-08-21:
Hi @mattncsu,

really sorry for the delayed reply. In case you are still experiencing this issue, the cards we use are SDHC and they have been formatted in FAT32 as you've said.

Did you change the jumper pins on your board? This could be an issue with the SD card module not being powered correctly. If you have a multimeter, could you check if the SD card has 3V3 between these pins: 
[image]

Here's where it is on the board physically:
[image]

amir, 2024-02-03:
Hi, apologies for reviving this old issue, but I'm experiencing the same problem. I have not touched the jumper pins on the board. If there are no SD cards in the module, I get a reading of about 70mV and almost zero with a SD card inserted.

RandalR, 2024-10-04:
Hi, I am having the same problem - was there any fix or resolution for this?  I've also not touched the jumpers and there is an 8GB SDHC FAT32 sd card in the reader.  I was trying the examples on my soldered_inkplate10 and can't get a few to work as I get an error similar to:

```
C:\foo>python pyboard.py --device COM3 exampleSd.py
Sd card could not be read
Traceback (most recent call last):
  File "<stdin>", line 21, in <module>
OSError: [Errno 2] ENOENT`
```
and
```
C:\foo>python pyboard.py --device COM3 drawGrayscaleImage.py
GS2: clean 1667ms (33ms ea), draw 1682ms (210ms ea), total 3349ms
Sd card could not be read
Starting to draw image from file!
Traceback (most recent call last):
  File "<stdin>", line 37, in <module>
  File "soldered_inkplate10.py", line 1086, in drawImageFile
OSError: [Errno 2] ENOENT

```

Please let me know what more I can try?  Thanks....

rsoric, 2024-10-07:
Hi @RandalR,

this is related to the MicroPython library, could you please open this as an issue there and I'll reply.

@amir , terribly sorry for missing your message here. Do you still need help with the SD card?

RandalR, 2024-10-07:
Sorry, I’m still new and finding my way around – where would you like me to file a bug report?

 

Thanks

 

              -Randal

amir, 2024-10-08:
@rsoric no worries at all, I'm still unable to use my SD card and have to load assets from a remote server which adversely impacts the battery life

rsoric, 2024-10-11:
@RandalR , we're still working on an official template for a bug report

@amir , glad you found a workaround in the meantime. I'm out of office for the time being but on Monday I'll be able to test this out on one of our boards.

One thing that comes to mind is that the SD card implementation with ESP32 is pretty straight forward, I think it's card compatibility that's the issue. Could you please try using the [official SD card formatter](https://www.sdcard.org/downloads/formatter/) to see if the card is in FAT32? Could you try a different SD card, Inoffice we are also using SD-XC cards, 8 and 16 gb.

RandalR, 2024-10-11:
Hi, I thought I had mentioned that I had read that maybe card compatibility may be an issue so that’s why I specifically used an older 8GB SanDisk.  I had also read in the docs that it would be looking for a FAT32 filesystem, which is what I put on it.  I can try with that formatter, but I seriously doubt that it will result in a different outcome.

 

I thought it was a hardware issue since I was reading about “jumpers” and voltages?

 

Thanks for your message and for your help

 

              -Randal R.

rsoric, 2024-10-14:
Hi @RandalR,

The reason I was mentioning those things is that it was my original thought when I first saw this ticket. From what I understand the boards weren't modified in any way via hardware - the microSD slot should work just fine. After consulting with my senior colleagues, there could be a potential issue with some Inkplate 10 boards regarding the SD card.

Could you please take a picture of the back of your board and also to identify the version number, printed faintly in one of the corners:
[image]

RandalR, 2024-10-14:
Thanks for getting back to me, here are the photos you requested.
[image]
[image]

Please let me know if you need a different view or if something isn't clear or visible.

Thanks for your help!

RandalR, 2024-10-14:
... a bit later I thought maybe a closer view of the back nearer the SD card would be more helpful, so here is that photo ... 
[image]

rsoric, 2024-10-15:
Hi @RandalR , thanks, that helps. Can you try shorting JP1 and seeing if that fixes the issue? This will ensure that the microSD card has constant power. If that doesn't help, we'll have to call it a hardware issue and will kindly ask you to contact us via [customer support ](https://soldered.com/contact/)to continue resolving this.

RandalR, 2024-10-15:
Thanks for your quick reply @rsoric  -  before I break out the soldering iron, I was wondering if there was a definitive test program that I could try?  Something that you know works in MicroPython so I can try it here?  I've only loaded the examples onto it for now - I think it was "exampleSd.py" - and that's where this issue popped up.

If that example should absolutely work I can then try the hardware jumper.     I just wanted to be sure it wasn't something that I did since I am not an experienced developer of these sorts of devices and it has been many years since I worked on an Arduino board...

rsoric, 2024-10-15:
Hey @RandalR , of course, I just tested this, i have placed the files from soldered_inkplate10/SD_card_example_files in the root folder of my SD card and I placed the SD card in the device and I ran this:

```
C:\repos\Inkplate-micropython>python pyboard.py --device COM10 -f cp soldered_inkplate10.py gfx.py gfx_standard_font_01.py mcp23017.py PCAL6416A.py image.py shapes.py soldered_logo.py :
cp soldered_inkplate10.py :soldered_inkplate10.py
cp gfx.py :gfx.py
cp gfx_standard_font_01.py :gfx_standard_font_01.py
cp mcp23017.py :mcp23017.py
cp PCAL6416A.py :PCAL6416A.py
cp image.py :image.py
cp shapes.py :shapes.py
cp soldered_logo.py :soldered_logo.py

C:\repos\Inkplate-micropython>python pyboard.py --device COM10 "Examples/soldered_inkplate10/exampleSd.py"
Sd card could not be read
Traceback (most recent call last):
  File "<stdin>", line 21, in <module>
OSError: [Errno 2] ENOENT

C:\repos\Inkplate-micropython>python pyboard.py --device COM10 "Examples/soldered_inkplate10/exampleSd.py"
Sd card could not be read
Traceback (most recent call last):
  File "<stdin>", line 21, in <module>
OSError: [Errno 2] ENOENT

C:\repos\Inkplate-micropython>python pyboard.py --device COM10 "Examples/soldered_inkplate10/exampleSd.py"
['System Volume Information', 'deepsleep_log.txt', '123', 'SCAN_00', '1', '2', '3', '4', 'random_file.txt, '1.bmp', 'text.txt']
Lorem ipsum dolor sit amet, consectetur adipiscing elit. Vivamus lacinia odio vitae vestibulum. Donec in efficitur leo, nec vehicula elit.

Suspendisse potenti. Nullam quis risus eget urna mollis ornare vel eu leo. Curabitur non nulla sit amet nisl tempus convallis quis ac lectus.

Praesent sapien massa, convallis a pellentesque nec, egestas non nisi. Pellentesque in ipsum id orci porta dapibus. Vestibulum ante ipsum primis in faucibus orci luctus et ultrices posuere cubilia curae.

Mauris blandit aliquet elit, eget tincidunt nibh pulvinar a. Vivamus suscipit tortor eget felis porttitor volutpat. Cras ultricies ligula sed magna dictum porta.

Quisque velit nisi, pretium ut lacinia in, elementum id enim. Vivamus magna justo, lacinia eget consectetur sed, convallis at tellus. Sed porttitor lectus nibh, et malesuada fames ac turpis egestas.

```

As you can see I could read from the .txt file and list all the files on the SD card. It took a while but the .bmp also drew:
[image]

The example should thus work.

RandalR, 2024-10-15:
@rsoric - Thanks for this.  Following your example I was able to get the image to load.   However, I think I've discovered maybe the root cause of my issue with the SD card::

Go through your example again (you can comment out the drawImage part for time's sake) and after the last line completes, run the same command again, just repeat the "exampleSd.py" program.

Here, if I do this, I get the error:
```
Sd card could not be read
Traceback (most recent call last):
  File "<stdin>", line 19, in <module>
OSError: [Errno 2] ENOENT
```

BUT if I power-cycle the device between runs (via the button next to the SD card slot), it works without giving an error.  But I **must** turn it off and back on to enable it to do so, thus:

```
2024-10-15 5:28:17.38
C:\InkplateMicroPython> python pyboard.py --device COM4 "mytestexamples\dir_inkplate_sd.py"
['System Volume Information', '1.bmp', 'text.txt']

## works fine

## run the same command again

2024-10-15 5:35:06.12 
C:\InkplateMicroPython> python pyboard.py --device COM4 "mytestexamples\dir_inkplate_sd.py"
Sd card could not be read
Traceback (most recent call last):
  File "<stdin>", line 19, in <module>
OSError: [Errno 2] ENOENT

## fails

## power cycle the board using the button next to the SD card slot

2024-10-15 5:35:15.61
C:\InkplateMicroPython> python pyboard.py --device COM4 "mytestexamples\dir_inkplate_sd.py"
['System Volume Information', '1.bmp', 'text.txt']

## works again

```

I am fairly certain that - being new to the platform - I was unaware of this particular requirement, and if it was mentioned anywhere in the documentation then I missed it.   This might have led to my frustration and conclusion that there was a problem or hardware issue.  Of course, if yours works fine, but mine continues to fail in this way, then there is still an issue.

Please let me know what you think and if you were able to duplicate my results.

rsoric, 2024-10-15:
Hey @RandalR , so I've got good news, I was able to replicate this issue and I've found the root cause. It's basically the main micropython firmware version.

First my hunch was to try and wait longer for the hardware reset of the SD card, and to unmount before trying to mount again - but this didn't fix the issue.

Then I was looking this up and found some users have fixed it by updating to a higher verison of micropython firmware, we use 1.18 included in this library, but versions up to 1.23 are available. This is running the same sketch (without the drawImage):

```
C:\repos\Inkplate-micropython>python pyboard.py --device COM10 "Examples/soldered_inkplate10/exampleSd.py"
['System Volume Information', 'deepsleep_log.txt', ...other files go here...]
Lorem ipsum dolor sit amet, consectetur adipiscing elit. Vivamus lacinia odio vitae vestibulum. Donec in efficitur leo, nec vehicula elit.

Suspendisse potenti. Nullam quis risus eget urna mollis ornare vel eu leo. Curabitur non nulla sit amet nisl tempus convallis quis ac lectus.

Praesent sapien massa, convallis a pellentesque nec, egestas non nisi. Pellentesque in ipsum id orci porta dapibus. Vestibulum ante ipsum primis in faucibus orci luctus et ultrices posuere cubilia curae.

Mauris blandit aliquet elit, eget tincidunt nibh pulvinar a. Vivamus suscipit tortor eget felis porttitor volutpat. Cras ultricies ligula sed magna dictum porta.

Quisque velit nisi, pretium ut lacinia in, elementum id enim. Vivamus magna justo, lacinia eget consectetur sed, convallis at tellus. Sed porttitor lectus nibh, et malesuada fames ac turpis egestas.
Traceback (most recent call last):
  File "<stdin>", line 40, in <module>
  File "soldered_inkplate10.py", line 880, in display
AttributeError: 'Inkplate' object has no attribute 'ipm' <-  This error happened because I had to remove the framebuffer so the script runs correctly, this is due to the differences between micropython versions

C:\repos\Inkplate-micropython>python pyboard.py --device COM10 "Examples/soldered_inkplate10/exampleSd.py"
['System Volume Information', 'deepsleep_log.txt', ...other files go here...]
Lorem ipsum dolor sit amet, consectetur adipiscing elit. Vivamus lacinia odio vitae vestibulum. Donec in efficitur leo, nec vehicula elit.

Suspendisse potenti. Nullam quis risus eget urna mollis ornare vel eu leo. Curabitur non nulla sit amet nisl tempus convallis quis ac lectus.

Praesent sapien massa, convallis a pellentesque nec, egestas non nisi. Pellentesque in ipsum id orci porta dapibus. Vestibulum ante ipsum primis in faucibus orci luctus et ultrices posuere cubilia curae.

Mauris blandit aliquet elit, eget tincidunt nibh pulvinar a. Vivamus suscipit tortor eget felis porttitor volutpat. Cras ultricies ligula sed magna dictum porta.

Quisque velit nisi, pretium ut lacinia in, elementum id enim. Vivamus magna justo, lacinia eget consectetur sed, convallis at tellus. Sed porttitor lectus nibh, et malesuada fames ac turpis egestas.
Traceback (most recent call last):
  File "<stdin>", line 40, in <module>
  File "soldered_inkplate10.py", line 880, in display
AttributeError: 'Inkplate' object has no attribute 'ipm'

C:\repos\Inkplate-micropython>
```

As you can see it works in this way. Unfortunately this isn't possible to fix in a simple manner, we would have to port the library to a newer version of the micropython firmware, but I'm glad we atleast found the cause.

rsoric, 2024-10-16:
@RandalR ,you're very welcome, I'm happy we got to a solution even though this ticket originally got derailed for a second (this was the solution for Micropython) 

@amir would it be possible for you to open a separate ticket with describing your issue?

@mattncsu are you still following the ticket?