SolderedElectronics/Inkplate-Arduino-library #202: Inkplate 6 color - Deep sleep issue

zBart, 2023-05-07:
I'm having some strange behavior with how my Inkplate6 color wakes up from deep sleep. Everytime it wakes up from deep sleep, it almost instantly does an additional reset, causing it to lose some data I'm storing in RTC.

Using the example: https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/master/examples/Inkplate6COLOR/Advanced/DeepSleep/Inkplate6COLOR_Wake_Up_Button/Inkplate6COLOR_Wake_Up_Button.ino and with the board powered via USB I get the following logs:
```
14:20:15.657 -> ets Jul 29 2019 12:21:46
14:20:15.657 -> 
14:20:15.657 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:20:15.657 -> configsip: 0, SPIWP:0xee
14:20:15.703 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:20:15.703 -> mode:DIO, clock div:1
14:20:15.703 -> load:0x3fff0030,len:1344
14:20:15.703 -> load:0x40078000,len:13864
14:20:15.703 -> load:0x40080400,len:3608
14:20:15.703 -> entry 0x400805f0
14:20:16.213 -> ets Jul 29 2019 12:21:46
14:20:16.213 -> 
14:20:16.213 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:20:16.213 -> configsip: 0, SPIWP:0xee
14:20:16.213 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:20:16.213 -> mode:DIO, clock div:1
14:20:16.213 -> load:0x3fff0030,len:1344
14:20:16.213 -> load:0x40078000,len:13864
14:20:16.213 -> load:0x40080400,len:3608
14:20:16.213 -> entry 0x400805f0

----

14:21:14.584 -> ets Jul 29 2019 12:21:46
14:21:14.584 -> 
14:21:14.584 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:21:14.584 -> configsip: 0, SPIWP:0xee
14:21:14.584 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:21:14.584 -> mode:DIO, clock div:1
14:21:14.584 -> load:0x3fff0030,len:1344
14:21:14.584 -> load:0x40078000,len:13864
14:21:14.584 -> load:0x40080400,len:3608
14:21:14.584 -> entry 0x400805f0
14:21:15.069 -> ets Jul 29 2019 12:21:46
14:21:15.069 -> 
14:21:15.069 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:21:15.112 -> configsip: 0, SPIWP:0xee
14:21:15.112 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:21:15.112 -> mode:DIO, clock div:1
14:21:15.112 -> load:0x3fff0030,len:1344
14:21:15.112 -> load:0x40078000,len:13864
14:21:15.112 -> load:0x40080400,len:3608
14:21:15.112 -> entry 0x400805f0
```

You can see the Inkplate wake up (`DEEPSLEEP_RESET`) and about 500ms later it resets again (`POWERON_RESET`). It happens again 30 seconds later (and keeps doing it). The screen also keeps displaying 'Boot count: 1'.

This happens both on wakeup from timer as well as with the wakeup from button.

To make this a more confusing, this only seems to happen on the new Inkplate6 color board (the one without touch pads on the front and the small bezel), I have two boards I recently got and it happens on both. The older board version (wider bezel / three touch pads on the front) works fine (both wake from timer and button).

Any clue on what is causing this?

zBart, 2023-05-21:
Small update on this, I did some more messing around and I noticed that I was actually still getting some logs between the `DEEPSLEEP_RESET` and the `POWERON_RESET`.

After placing some `Serial.println()` calls around I actually narrowed it down to around the `memset()` call here:

https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/master/src/boards/Inkplate6Color.cpp#L64
```cpp
        // Allocate memory for internal frame buffer
        DMemory4Bit = (uint8_t *)ps_malloc(E_INK_WIDTH * E_INK_HEIGHT / 2);
        if (DMemory4Bit == NULL)
        {
            return false;
        }
		
        Serial.println("pre-memset");

        // Color whole frame buffer in white color
        memset(DMemory4Bit, INKPLATE_WHITE | (INKPLATE_WHITE << 4), E_INK_WIDTH * E_INK_HEIGHT / 2);
    }
	
    Serial.println("pre-resetpanel");

    // Reset epaper
    resetPanel();
```

With the debug println's added by me, the log now shows:
```
16:19:53.393 -> ets Jul 29 2019 12:21:46
16:19:53.393 -> 
16:19:53.393 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:19:53.393 -> configsip: 0, SPIWP:0xee
16:19:53.393 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:19:53.439 -> mode:DIO, clock div:1
16:19:53.439 -> load:0x3fff0030,len:1344
16:19:53.439 -> load:0x40078000,len:13864
16:19:53.439 -> load:0x40080400,len:3608
16:19:53.439 -> entry 0x400805f0
16:19:58.924 -> pre-memset
16:19:58.924 -> pre-resetp�ets Jul 29 2019 12:21:46
16:19:58.924 -> 
16:19:58.924 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:19:58.924 -> configsip: 0, SPIWP:0xee
16:19:58.924 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:19:58.924 -> mode:DIO, clock div:1
16:19:58.924 -> load:0x3fff0030,len:1344
16:19:58.969 -> load:0x40078000,len:13864
16:19:58.969 -> load:0x40080400,len:3608
16:19:58.969 -> entry 0x400805f0
16:20:04.559 -> pre-memset
16:20:04.559 -> pre-resetpanel
```

So to me it seems that shortly after the `memset()`, the board resets.

To see if it some kind of timing thing, I added a `vTaskDelay(10000);` before the `memset`, but the issue persists:
```
16:25:20.490 -> ets Jul 29 2019 12:21:46
16:25:20.490 -> 
16:25:20.490 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:25:20.490 -> configsip: 0, SPIWP:0xee
16:25:20.490 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:25:20.535 -> mode:DIO, clock div:1
16:25:20.535 -> load:0x3fff0030,len:1344
16:25:20.535 -> load:0x40078000,len:13864
16:25:20.535 -> load:0x40080400,len:3608
16:25:20.535 -> entry 0x400805f0
16:25:21.032 -> pre-taskdelay
16:25:31.014 -> pre-memset
16:25:31.014 -> pre-res�ets Jul 29 2019 12:21:46
16:25:31.014 -> 
16:25:31.014 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:25:31.014 -> configsip: 0, SPIWP:0xee
16:25:31.014 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:25:31.059 -> mode:DIO, clock div:1
16:25:31.059 -> load:0x3fff0030,len:1344
16:25:31.059 -> load:0x40078000,len:13864
16:25:31.059 -> load:0x40080400,len:3608
16:25:31.059 -> entry 0x400805f0
16:25:31.652 -> pre-taskdelay
16:25:41.648 -> pre-memset
16:25:41.648 -> pre-resetpanel
```

So you can see the reset still happens right after the memset.

rsoric, 2023-08-21:
Hi @zBart 

really sorry for the late reply, are you still experiencing this issue?

Are you using the correct board definition for your Inkplate? The new Inkplate is "Soldered Inkplate6COLOR" in Arduino IDE. The begin function should run after wakeup all over again because of how Deep Sleep on the ESP works. 

One thing that comes to mind is that it's a reset due to low current from the power supply. Could you try this with a different power supply which outputs a higher amperage and see if the problem still persists?

Vdauphin, 2023-09-12:
Hello here,
I am experiencing the same issue, mainly using https://github.com/SolderedElectronics/Inkplate-Arduino-library/tree/master/examples/Inkplate6COLOR/Advanced/DeepSleep/Inkplate6COLOR_Simple_Deep_Sleep
But changed to `#define TIME_TO_SLEEP  2 * 60` and added some `Serial.print`:
```c
/*
   Inkplate6COLOR_Simple_Deep_Sleep  example for Soldered Inkplate 6COLOR
   For this example you will need USB cable and Inkplate 6COLOR.
   Select "Soldered Inkplate 6COLOR" from Tools -> Board menu.
   Don't have "Soldered Inkplate 6COLOR" option? Follow our tutorial and add it:
   https://soldered.com/learn/add-inkplate-6-board-definition-to-arduino-ide/

   This example will show you how you can use low power functionality of Inkplate board.
   In deep sleep, whole board will consume about 25uA from battery.
   Inkplate will wake every 20 seconds change content on screen.

   NOTE: Because we are using deep sleep, everytime the board wakes up, it starts program from begining.
   Also, whole content from RAM gets erased, so you CAN NOT use partial updates.

   Want to learn more about Inkplate? Visit www.inkplate.io
   Looking to get support? Write on our forums: https://forum.soldered.com/
   15 July 2020 by Soldered
*/

// Next 3 lines are a precaution, you can ignore those, and the example would also work without them
#ifndef ARDUINO_INKPLATECOLOR
#error "Wrong board selection for this example, please select Soldered Inkplate 6COLOR in the boards menu."
#endif

#include "Inkplate.h"      //Include Inkplate library to the sketch
#include "driver/rtc_io.h" //ESP32 library used for deep sleep and RTC wake up pins
#include "picture1.h" //Include .h files of 3 pictures. All three pictures were converted using Inkplate Image Converter
#include "picture2.h"
#include "picture3.h"
const uint8_t *pictures[] = {picture1, picture2,
                             picture3}; // This array of pinters holds address of every picture in the memory,
                                        // so we can easly select it by selecting index in array

#define uS_TO_S_FACTOR 1000000 // Conversion factor for micro seconds to seconds
#define TIME_TO_SLEEP  2 * 60      // How long ESP32 will be in deep sleep (in seconds)
RTC_DATA_ATTR int slide = 0;

Inkplate display; // Create an object on Inkplate library and also set library into 3 Bit mode (gray)

void setup()
{
    Serial.begin(115200);
    Serial.println("Booting");
    Serial.print("display.begin() ");
    Serial.println(display.begin()); // Init Inkplate library (you should call this function ONLY ONCE)
    display.clearDisplay(); // Clear frame buffer of display
    display.drawImage(
        pictures[slide], 0, 0, 600,
        448); // Display selected picture at location X=0, Y=0. All three pictures have resolution of 600x448 pixels
    display.display(); // Refresh the screen with new picture
    slide++; // Update counter for pictures. With this variable, we choose what picture is going to be displayed on
             // screen
    if (slide > 2)
        slide = 0; // We do not have more than 3 images, so roll back to zero

    // Uncomment this line if your Inkplate is older than Aug 2021 as older Inkplates have ESP32 wrover-e chips
    // rtc_gpio_isolate(GPIO_NUM_12); // Isolate/disable GPIO12 on ESP32 (only to reduce power consumption in sleep)

    // Activate wake-up timer -- wake up after 20s here
    esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);

    // Put ESP32 into deep sleep. Program stops here.
    esp_deep_sleep_start();
}

void loop()
{
    // Nothing! If you use deep sleep, whole program should be in setup() because each time the board restarts, not in a
    // loop()! loop() must be empty!
}

```

After two minutes, the display get clear and still show the first picture, the console show:
```c
21:44:48.251 -> ets Jul 29 2019 12:21:46
21:44:48.251 -> 
21:44:48.251 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:44:48.251 -> configsip: 0, SPIWP:0xee
21:44:48.251 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:44:48.251 -> mode:DIO, clock div:1
21:44:48.324 -> load:0x3fff0030,len:1344
21:44:48.324 -> load:0x40078000,len:13864
21:44:48.324 -> load:0x40080400,len:3608
21:44:48.324 -> entry 0x400805f0
21:44:48.766 -> Booting
21:44:48.766 -> display.begin() ets Jul 29 2019 12:21:46
21:44:48.766 -> 
21:44:48.766 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:44:48.766 -> configsip: 0, SPIWP:0xee
21:44:48.798 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:44:48.798 -> mode:DIO, clock div:1
21:44:48.798 -> load:0x3fff0030,len:1344
21:44:48.798 -> load:0x40078000,len:13864
21:44:48.798 -> load:0x40080400,len:3608
21:44:48.798 -> entry 0x400805f0
21:44:49.505 -> Booting
21:44:49.505 -> display.begin() 1
21:47:16.518 -> ets Jul 29 2019 12:21:46
21:47:16.518 -> 
21:47:16.518 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:47:16.550 -> configsip: 0, SPIWP:0xee
21:47:16.550 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:47:16.550 -> mode:DIO, clock div:1
21:47:16.550 -> load:0x3fff0030,len:1344
21:47:16.550 -> load:0x40078000,len:13864
21:47:16.550 -> load:0x40080400,len:3608
21:47:16.550 -> entry 0x400805f0
21:47:17.043 -> Booting
21:47:17.043 -> display.begin() ets Jul 29 2019 12:21:46
21:47:17.065 -> 
21:47:17.065 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:47:17.065 -> configsip: 0, SPIWP:0xee
21:47:17.065 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:47:17.065 -> mode:DIO, clock div:1
21:47:17.065 -> load:0x3fff0030,len:1344
21:47:17.065 -> load:0x40078000,len:13864
21:47:17.065 -> load:0x40080400,len:3608
21:47:17.065 -> entry 0x400805f0
21:47:17.770 -> Booting
21:47:17.770 -> display.begin() 1
```

Looks like the `display.begin()` is crashing the device

Any input on this?

rsoric, 2023-09-13:
@Vdauphin , Thanks for providing more details. I've just tried to recreate your Issue here and I couldn't get Inkplate 6COLOR to crash. I used the exact same code you posted and IP6 color (both the blue PCB and purple PCB version) wake from sleep just fine. 

I suggest we try the following, as I can't recreate your issue I'm going to need some more co-operation in order to truly determine what's causing this and to fix the issue: 

1) Could you please use the ESP exception decoder and see what exactly the stack trace tells you? Here's a link to it: https://github.com/me-no-dev/EspExceptionDecoder - this is a tool which is really useful as it will pinpoint the exact function call which is crashing the device. Installation instructions for it and tutorial is in the README. Write back with what the exception decoder tells you.

2) After that, could you try and download the latest version of the Inkplate library from the /dev branch (instead of main), try the compiling example again and see if it works.

Vdauphin, 2023-09-13:
Hello,
Thank you for your fast reply, I don't have that much knowledge on this. I will try my best
1. I installed it and follow steps in the README but I don't see any exception popping in my serial monitor. What I my missing?

rsoric, 2023-09-14:
@Vdauphin Ah, so the ESP seems to be restarting OK, glad we could rule that out.

I've created a modified Inkplate6COLOR.cpp file: https://pastebin.com/raw/W3UFGKA6
I've added some calls to Serial.print so we can better see where exactly display.begin crashes. 

Use this file and paste it in `src/boards/Inkplate6Color.cpp` in your Inkplate library locally on your computer. 

Let me know what this prints out, again. Sorry to have to do this in this way but I can't replicate the issue locally.

Vdauphin, 2023-09-14:
Here is the output :)

```c
19:58:28.070 -> ets Jul 29 2019 12:21:46
19:58:28.070 -> 
19:58:28.070 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:58:28.070 -> configsip: 0, SPIWP:0xee
19:58:28.070 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:58:28.070 -> mode:DIO, clock div:1
19:58:28.070 -> load:0x3fff0030,len:1344
19:58:28.070 -> load:0x40078000,len:13864
19:58:28.103 -> load:0x40080400,len:3608
19:58:28.103 -> entry 0x400805f0
19:58:28.586 -> Booting
19:58:28.586 -> display.begin() Inside display.begin()!
19:58:28.586 -> _beginDone is false!
19:58:28.586 -> Just after allocating memory for framets Jul 29 2019 12:21:46
19:58:28.586 -> 
19:58:28.586 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:58:28.586 -> configsip: 0, SPIWP:0xee
19:58:28.586 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:58:28.586 -> mode:DIO, clock div:1
19:58:28.619 -> load:0x3fff0030,len:1344
19:58:28.619 -> load:0x40078000,len:13864
19:58:28.619 -> load:0x40080400,len:3608
19:58:28.619 -> entry 0x400805f0
19:58:29.294 -> Booting
19:58:29.294 -> display.begin() Inside display.begin()!
19:58:29.294 -> _beginDone is false!
19:58:29.294 -> Just after allocating memory for frame buffer
19:58:29.326 -> Just before waking the ePaper and initing
19:58:29.520 -> Just before putting the panel back to sleep!
19:58:29.616 -> 1
```

rsoric, 2023-09-18:
Thanks! So, it seems that upon the first deep sleep reset, the crash happens shortly after allocating the frame buffer memory. This should definitely happen upon restart as this memory is dynamic and not held in the RTC either. 

I'm guessing this could be a power spike which is drawing too much current causing a restart. I would try powering Inkplate over a stronger USB power supply and seeing if it still double-restarts. 

I'm going to be looking into this with more detail tomorrow to provide a more specific answer so hang tight :)

rsoric, 2023-09-21:
@Vdauphin 

After consulting with a colleague we determined that you may need to add a 10k pull-up resistor to the manual reset line, you may have one of the earlier designs. It seems that the lack of this resistor is causing Inkplate to reset when it's not supposed to. 

To best guide you where to add this resistor exactly to try and fix this, could you please let me know of the exact version of the Inkplate6COLOR board you have? Look at the corners of the PCB, they sould have a version printed (like 1.2.0). If you can't find it, please send a photo of the back of your Inkplate and we'll figure out the version from there.

Vdauphin, 2023-09-21:
> Look at the corners of the PCB, they sould have a version printed (like 1.2.0)

Yes it is 1.2.0

Vdauphin, 2023-09-25:
It doesn't change anything
I have 0.78V between the 10k resistor pin:
[image]

rsoric, 2023-09-26:
@Vdauphin thanks! My colleague claims that this should be 3v3 so this is strange behavior. Can you measure 3V3/GND and VIN/GND?

Vdauphin, 2023-09-26:
May be it was unclear, when the USB is not pluging:
- 3V3/GND: 3.3V
- VIN/GND: 3.72V
- green circle/GND: 3.3V
- green circle/3V3: 0V

rsoric, 2023-09-28:
@Vdauphin Thanks for letting us know, manual reset then seems fine. Please contact us over [customer support](https://soldered.com/contact/) as this seems like a potential hardware issue, we can't replicate it on the boards that we have - we will help you out from there. Just link this GitHub page in your ticket and we'll sort it out for you.

zBart, 2023-10-08:
Hi @rsoric, 

I finally had some time to spend on this again. I did some experimentation and found out this (based on the Inkplate6COLOR_Wake_Up_Button example):
-

zBart, 2023-10-08:
Hi @rsoric, 

I finally had some time to spend on this again. 

I did some experimentation and found out this (based on the `Inkplate6COLOR_Wake_Up_Button` example):

- On USB - wake up button - works fine
- On USB - timer wake up - always seems to have issue
- On battery - wake up button - works fine
- On battery - timer wake up - seems to be working now, even though it previously didn't... Maybe batter power related?

Here is the log, on USB with timer wake up, with your debug statements from https://pastebin.com/raw/W3UFGKA6:
```
22:08:49.317 -> ets Jul 29 2019 12:21:46
22:08:49.317 -> 
22:08:49.317 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:08:49.317 -> configsip: 0, SPIWP:0xee
22:08:49.317 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:08:49.360 -> mode:DIO, clock div:1
22:08:49.360 -> load:0x3fff0030,len:1344
22:08:49.360 -> load:0x40078000,len:13864
22:08:49.360 -> load:0x40080400,len:3608
22:08:49.360 -> entry 0x400805f0
22:08:49.831 -> Booting
22:08:49.831 -> Inside display.begin()!
22:08:49.831 -> _beginDone is false!
22:08:49.831 -> Just after allocating memory for frame buffer
22:08:49.831 -> Just before�ets Jul 29 2019 12:21:46
22:08:49.875 -> 
22:08:49.875 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:08:49.875 -> configsip: 0, SPIWP:0xee
22:08:49.875 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:08:49.875 -> mode:DIO, clock div:1
22:08:49.875 -> load:0x3fff0030,len:1344
22:08:49.875 -> load:0x40078000,len:13864
22:08:49.875 -> load:0x40080400,len:3608
22:08:49.875 -> entry 0x400805f0
22:08:50.473 -> Booting
22:08:50.473 -> Inside display.begin()!
22:08:50.473 -> _beginDone is false!
22:08:50.473 -> Just after allocating memory for frame buffer
22:08:50.473 -> Just before waking the ePaper and initing
22:08:50.675 -> Just before putting the panel back to sleep!
```

I can't find a version number on my board.

rsoric, 2023-10-09:
Hi @zBart 

Glad you wrote back, and thanks for writing this comprehensive info for us. As you can see, it crashes in the exact same time as @Vdauphin 's  output. It occurred to me that this could potentially be an issue with the core for ESP32 which we're building upon. We tried to exhaust all other options with testing remotely.

For now, I recommend you reach us through customer support to try and resolve this - but I'll write back here with some more info after I bring this up to our team.

rsoric, 2023-10-26:
Hi @zBart and @Vdauphin , 

good news - we have figured out what was causing this. Some pinModes needed to be set during when waking the panel from deep sleep, and another configuration registry entry needed to be done which was being skipped. So, it's a driver issue.

Please give [this ](https://pastebin.com/raw/jry6fPEA)code a try for Inkplate6Color.cpp.

I just tested it on a new version of the device and it resets only once with DEEPSLEEP_RESET.

Vdauphin, 2023-10-26:
Hello, 
Oh okay! Sorry but I already sent my device to you
Cheers

rsoric, 2023-10-27:
@Vdauphin  Not to worry, regarding your support ticket, your replacement device which has has this feature tested is on it's way.

rsoric, 2023-10-30:
Closing this issue as a push to the main branch was made which fixes this, see details [here](https://github.com/SolderedElectronics/Inkplate-Arduino-library/commit/cbb736fc72e2adddcfdcfa80cec47b7029a8c8ae#diff-3954d35932e464622d3bf6e9460d6b36dd45e922443404279c4ca8c9e7d82fa6). 

@Vdauphin and @zBart thanks again for working with us to help resolve this for all other users.

closed 2023-10-30 by rsoric