SolderedElectronics/Inkplate-Arduino-library #203: Possible duplicate of issue #166: Was issue 166 ever resolved?

petitchou, 2023-05-15:
Hi there!

I am trying to compile for Inkplate 10 and am running into [this same issue](https://github.com/SolderedElectronics/Inkplate-Arduino-library/issues/166) that another user was. A fix was promised "in a few days," but it has obviously been quite some time since this issue was submitted. I pulled down the 1.0.1 version of the Inkplate drivers straight from the Arduino IDE following the instructions provided. I can't tell if there's something I need to be doing that I haven't done, if the instructions are out of date, or if the fix was just never made. Could you please advise? :)

Thanks so much!

rsoric, 2023-05-19:
Hi @petitchou 

You should get the board definitions from here https://inkplate.readthedocs.io/en/latest/get-started.html#arduino and the Arduino library from here: https://github.com/SolderedElectronics/Inkplate-Arduino-library

Get them from GitHub and not from the Arduino IDE, this will ensure you have the latest, most updated version and getting them from GitHub is how we recommend users to install which can be seen in our [docs](https://inkplate.readthedocs.io/en/latest/get-started.html#arduino). From looking at the main branch of the Inkplate Library there is a function specifically for downloading files via HTTPS which uses the WiFiClientSecure class' function setInsecure, as seen [here ](https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/b713faa8e8bbc5f3deb17024a2671822a99918b5/src/include/NetworkClient.cpp#L185).

So, this should compile fine on our end, which you can see by the fact that the main branch passed all the compile tests we have on GitHub. 

So, I would recommend installing the latest board definitions from GitHub and also the Inkplate library. Make sure to delete the old versions. Also, for testing, we most often use Arduino IDE legacy version (1.8) so we would recommend trying with that as well. 

Let me know if you can get it to work this way, if not, write back and we will figure it out.

-Rob

rsoric, 2023-05-31:
Hi @olalid 

Thanks for pointing this out. We're going to update it.

-Rob