SolderedElectronics/Inkplate-Arduino-library #204: Inkplate10 board without touchpads

matthewmmorrow, 2023-05-25:
Just got an Inkplate10 today and it doesn't have touchpads. Is this a new version that should have its own board in Arduino?

I uploaded the wake on touchpads example and didn't realize they were missing until I opened up the case! Going to try to find a version with touchpads.

rsoric, 2023-05-26:
Hi @matthewmmorrow 

If you have a newer version of Inkplate 10, the one on the purple PCB - the correct board definition to use is "Soldered Inkplate 10". You can get the board definition for it here: https://github.com/SolderedElectronics/Dasduino-Board-Definitions-for-Arduino-IDE . The older one (blue pcb) has a board definition labeled with "e-Radionica Inkplate 10".

matthewmmorrow, 2023-05-26:
Do all the purple Inkplate 10 not have touchpads?

Should `readTouchpad()` be excluded for devices without touchpads? (For some reason it doesn't exist for the v2 6Color even though it has touchpads)

rsoric, 2023-05-29:
Hi,

I have consulted with a colleague about this. There is, in fact, a smaller batch of Inkplates that are purple and have touchpads. 

The best way to determine which board you have between the older e-Radionica ones (V1) or the newer Soldered ones (V2) is by using this guide from our docs: https://inkplate.readthedocs.io/en/latest/arduino.html#io-expander-functions

As you can see, it mainly has to do with the GPIO expander.

As for `readTouchpad()` being excluded if the actual board doesn't have touchpads - we found that using too many `#define` macros makes things more complicated really quickly - and we want to make our examples easy to read and use, so, for now the function is visible on all boards which have a version with touchpads.

matthewmmorrow, 2023-05-30:
Got it, I definitely have one of the new purple ones without touchpads. As to the point of the issue, it seems that there are 3 different hardware versions of the board and only 2 different board definitions. Both purple Soldered boards, with or without touchpads, should use the V2 board definition.

closed 2023-05-31 by rsoric