SolderedElectronics/Inkplate-Arduino-library #207: Inkplate MCP23017 Expander not working

K-J-Q, 2023-06-17:
Tried running the example, but I did not get any response from GPB7 pin. Measured using the multimeter - reading 0 V. 

EInk display still works.

rsoric, 2023-06-20:
Hi @K-J-Q 

Wich Inkplate do you have? You might be aware that the newer ones have a different GPIO expander. Check which one you have by using the information on this link: https://inkplate.readthedocs.io/en/latest/arduino.html?highlight=gpio%20expander#io-expander-functions 

We made the library to be agnostic of this - and if the ePaper screen updates, that means you most likely have the correct board definition for your Inkplate selected, but it's worth checking.

Does the GPIO expander example work on any other pins? Are you using the internal or external GPIO expander?

Please fill me in with some more info so I can help you take care of this issue.

-Rob

K-J-Q, 2023-06-21:
@rsoric,

I am using the older version, MCP. I am testing on the internal GPIO expander pins, but it dosen't work on any of the B ports. Strangely, the eink display and touchpads still works. 

The pins were tested by writing HIGH, and measuring the voltage using a multimeter.

rsoric, 2023-06-26:
That's strange, this could potentially be a hardware issue. Let's try to rule that out first. From what I see on the schematic, the eInk display and touchpads are connnected to the first MCP. The second MCP could then have a bad connection somewhere. 

You still haven't let me know which version of Inkplate you have, I'm going to assume it's e-Radionica Inkplate 10. Here is the schematic for how the second MCP is connected: 

[image]

Here is the pinout of the MCP:
[image]

Please check with your multimeter the I2C SCL and SDA connections, with VCC and GND - they should all be connected with those pins elsewhere on the Inkplate.

K-J-Q, 2023-06-27:
Looks correct. Voltages and I2C pins are connected. BTW, I also soldered on another MCP23017 chip. Not sure if that will affect pinouts.

rsoric, 2023-06-28:
What do you mean you soldered on another MCP23017, have you replaced the one that came with the board or have you placed another extra one somewhere?

I'm thinking it could be an issue with I2C address overlap. The second MCP should show up as I2C address 0x22 or 0x21 (depending on which version you have). Please run the I2C [scanner sketch](https://gist.github.com/walidamriou/1e759043d66340f9c76c7d13b6abf55e) and see if these addresses show up. More devices will show, but we are mainly interested in 0x22 or 0x21. 0x20 will show up for sure I think, that is the first GPIO expander. Let me know the results from this and we'll see from there.

K-J-Q, 2023-07-01:
My board comes with one MCP chip, due to the chip shortage. I decided to purchase a MCP23017 chip and solder it onto the PCB (MCP23017-2). 

I2C detected 0x20, but not 0x21/0x22. Somehow, the second MCP is not detected. 

Would adding the second expander chip (MCP23017-2) cause the first expander chip (MCP23017-1) chip to not work correctly?

rsoric, 2023-07-03:
@K-J-Q , now I understand. So, the second MCP is potentially not detected either because of a bad soldering connection or a short. The only way it would not get detected is if it's on the same address as the first one or isn't connected well.

Could you check the address select pins, A2, A1 and A0? To make sure the second MCP is running on a different address, connect one of the pins to 3V3, check if it's connected in this way with your multimeter. Here is the chart which describes the address assignment using the hardware address pins: 
[image]

K-J-Q, 2023-07-08:
@rsoric The second MCP chip was not being detected and I desoldered it. The first MCP (MCP23017-1) still does not work when I run the example code. Could you test the example program on your side to verify that it is working?

rsoric, 2023-07-13:
@K-J-Q 

I was able to blink a LED on using the internal_io_expander example for Inkplate 10 just fine

K-J-Q, 2023-07-16:
@rsoric Thanks for testing it out. Turns out it is because of the library i was using. It worked after installing the latest library (v8.0.0) :) I was using v5.7.1 before.

closed 2023-07-16 by K-J-Q