SolderedElectronics/Inkplate-Arduino-library #209: High deep sleep current draw with SD card on Inkplate 10 (v1)

chrisjtwomey, 2023-06-26:
I have a [PPK2](https://www.nordicsemi.com/Products/Development-hardware/Power-Profiler-Kit-2) and was testing deep sleep current draw and noticed that current draw is very high when the SD card is initialized:

With `display.sdCardInit()` - about **2mA**:
[image]

Without - about **110μA**:
[image]

This is the current setup (I turned off the power LED in my tests)
[image]

I see in the library there is a `sdCardSleep()` function that does not appear to have an affect on power consumption, I'm guessing because I have the old v1 version of the Inkplate10 and maybe I'm missing some hardware that helps turn off this power during deep sleep? Is there any alternative for the old version? I'd like to keep using the SD card module without it destroying battery life.

Also without the SD card module 100uA sleep current is good but nowhere near the 22uA power consumption mentioned in the example sketches. Is this exclusive to the newer versions of Inkplate10 too?

Sketch I'm using to test this:
```
#define uS_TO_S_FACTOR 1000000
#define mS_TO_S_FACTOR 1000
#define TIME_TO_WAKE_OR_SLEEP  2

#include "Inkplate.h"
Inkplate display(INKPLATE_3BIT);

void setup()
{
    display.begin();
 
    display.sdCardInit();
    delay(TIME_TO_WAKE_OR_SLEEP * mS_TO_S_FACTOR);
    esp_sleep_enable_timer_wakeup(TIME_TO_WAKE_OR_SLEEP * uS_TO_S_FACTOR);
    // display.sdCardSleep();

    esp_deep_sleep_start();
}

void loop(){}
```

rsoric, 2023-06-27:
Hi @chrisjtwomey 

having a look at the schematics side by side, the new version of Inkplate 10 has a MOSFET that has a GPIO pin connected which can enable or disable the power to the SD card. Here is the function definition of sdCardSleep:
[image]

As you can see, it's only supported by Inkplate 10 V2 and not V1. If you want, I could write you a short guide on adding this MOSFET to the board yourself, it would only take a few components. Here is the schematic:

[image]

chrisjtwomey, 2023-06-27:
> If you want, I could write you a short guide on adding this MOSFET to the board yourself, it would only take a few components. 

@rsoric Ok that would be great, thanks!

chrisjtwomey, 2023-06-27:
> Also without the SD card module 100uA sleep current is good but nowhere near the 22uA power consumption mentioned in the example sketches. 

For anyone that finds themselves in a similar situation, keeping the SD card in the holder consumes about **~85μA** in deep sleep, even if `sdCardInit()` was not called. Removing the SD card from the tray and the deep sleep current returns to **24μA**. This may only apply to the old e-radionica version, don't have the new version to test with.

rsoric, 2023-06-28:
@chrisjtwomey So, I initially had the idea of unsoldering the 3V3 pin on the SD card holder and attaching the MOSFET as is shown in the schematic I showed you, but the connection seems really close to the other pins and to the board. Seems to me that attempting this would be more trouble than it's worth.

I do have another proposition - how about using a SD card adapter and connecting it to the SPI pins? Something like [this ](https://soldered.com/product/microsd-breakout/). You could also attach the MOSFET between the VCC pin and 3v3, and use an additional GPIO pin to turn it on or off. I think this solution would be simpler to implement and would keep your board largely intact.

chrisjtwomey, 2023-06-29:
@rsoric Thanks for the info. Think I have an SD card breakout board somewhere I could give it a test at some point. For now in my current project I've gated the use of SD cards to just the V2 Inkplates. I'll experiment when I have a second inkplate perhaps 😅 

Will close this issue as my problem has been resolved for both SD cards and without - cheers!

closed 2023-06-29 by chrisjtwomey