SolderedElectronics/Inkplate-Arduino-library #212: Readme is oudated regarding Linux extra steps

robertoash, 2023-07-17:
The Readme documentation, in the section [Linux?](https://github.com/SolderedElectronics/Inkplate-Arduino-library#linux), says that these are the steps to follow:
```
apt install python3-pip
pip3 install pyserial
apt install python-is-python3
```

However, I got an error when trying to follow those instructions. It seems Debian and Ubuntu (and potentially Fedora?) are not allowing the use of pip to directly install packages onto the system Python installation to make it less likely that one of those packages might break the main system installation. Here's the article I read about it: https://www.linuxuprising.com/2023/03/next-debianubuntu-releases-will-likely.html

Doing some more research I found a solution that worked for me and allows me to compile my sketches throught the Arduino IDE:

```
apt install python3-serial
```

rsoric, 2023-07-19:
Hi @robertoash 

Thanks for letting us know this! We don't mainly use Linux so we didn't know about this - it will be corrected in an update to the README on the next official release of the library.

-Rob

closed 2023-07-19 by rsoric

rsoric, 2023-07-20:
[Here ](https://github.com/SolderedElectronics/Inkplate-Arduino-library/commit/0028bcc2d2a3dd7108a810e8adf7a7e88fce0a0e)is the commit which contains the info you provided us, it will end up on the main branch over time, thanks again.