SolderedElectronics/Inkplate-Arduino-library #214: Using Inkplate library with ESPAsyncWebServer breaks compilation

dkbay, 2023-08-03:
Hey!

When I try to includeb oth inkplate library and ESPAsyncWebServer I cannot compile in Arduino IDE.
However if I remove either library it works again.

It throws this error.
```
In file included from /Users/lukasb.kalkhoran/Documents/Arduino/libraries/ESP_Async_WebServer/src/ESPAsyncWebServer.h:27,
                 from /private/var/folders/ym/dbscpwz54594zltrf6dt2vcc0000gp/T/.arduinoIDE-unsaved202373-32347-1m5z0wv.16bt/sketch_aug3a/sketch_aug3a.ino:7:
/Users/lukasb.kalkhoran/Library/Arduino15/packages/Inkplate_Boards/hardware/esp32/3.0.1/libraries/FS/src/FS.h:120:11: error: 'class fs::File' conflicts with a previous declaration
 using fs::File;
           ^~~~
In file included from /Users/lukasb.kalkhoran/Documents/Arduino/libraries/InkplateLibrary/src/include/../libs/SdFat/FatLib/FatLib.h:27,
                 from /Users/lukasb.kalkhoran/Documents/Arduino/libraries/InkplateLibrary/src/include/../libs/SdFat/SdFat.h:33,
                 from /Users/lukasb.kalkhoran/Documents/Arduino/libraries/InkplateLibrary/src/include/Image.h:23,
                 from /Users/lukasb.kalkhoran/Documents/Arduino/libraries/InkplateLibrary/src/include/Graphics.h:24,
                 from /Users/lukasb.kalkhoran/Documents/Arduino/libraries/InkplateLibrary/src/Inkplate.h:26,
                 from /private/var/folders/ym/dbscpwz54594zltrf6dt2vcc0000gp/T/.arduinoIDE-unsaved202373-32347-1m5z0wv.16bt/sketch_aug3a/sketch_aug3a.ino:2:
/Users/lukasb.kalkhoran/Documents/Arduino/libraries/InkplateLibrary/src/include/../libs/SdFat/FatLib/ArduinoFiles.h:122:7: note: previous declaration 'class File'
 class File : public FatFile, public Stream {
       ^~~~

exit status 1

Compilation error: exit status 1
 ```
 
 It would be quite nice to be able to use both these libraries together!

rsoric, 2023-08-04:
Hi @dkbay 

So, it's a conflict within the SdFat library we're including. You essentially have two options:

- Modify the SdFat library so the class isn't called 'File' (and accordingly modify the entire Inkplate library)
- Modify the ESPAsyncWebServer library 

The latter seems simpler and more straight forward as it could be handled with a single case-sensitive find and replace on the level of the code library (VSCode can do this, for example). Rename the 'File' class to something like 'FileESPAsync' just to clarify to the compiler that it's a different class.

On Windows, libraries are installed under 'my documents/Arduino/libraries', check where this is on your operating system.

I'm telling you how to do this as we weren't planning on changing the class names for our included libraries. But it shouldn't be a problem to get this to work.

Note that if you update the ESPAsyncWebServer library you will have to do this find-and-replace again.
 
-Rob

dkbay, 2023-08-04:
Ah I see!

Thanks for the response Rob, sry for the bother I'm not too experienced with C/C++. 
Initially thought libraries would be scoped like NPM packages 😅 

I will try this out though!

Thanks again! :D

rsoric, 2023-08-04:
No problem, we run into similar things when making projects. As I've said, it should be a simple find-and-replace operation. Let me know if you get this to work.