SolderedElectronics/Inkplate-Arduino-library #223: Display Doesn't Work for Inkplate 10

dbreunig, 2023-10-14:
All set up with Arduino. Can print to serial, read from SD card...but any calls to display do nothing.

And since "Debugging is not supported by Soldered Inkplate 10" I can't really figure out what's happening. Any ideas for further debugging or fixes?

dbreunig, 2023-10-14:
Here's an example of what's being called:

```
#include "Inkplate.h"

Inkplate display(INKPLATE_3BIT);

void setup() {
  display.begin();

  // Start up a Serial monitor @ 115200 bauds.
  Serial.begin(115200);
  Serial.println("Start");

  Serial.println(display.getDisplayMode());
  display.setPanelState(1);
  Serial.println(display.getPanelState());

  display.clearDisplay();
  display.display();

  if (display.sdCardInit()) {
    Serial.println("SD Card OK");
  } else {
    Serial.println("SD Card not OK");
  }
}
```

This yields:

```
Start
1
1
SD Card OK
```

The display is not cleared.

rsoric, 2023-10-16:
Hi @dbreunig , I believe this is the same issue as the last one you've opened, and that these aren't two separate faults. Please check my reply [there](https://github.com/SolderedElectronics/Dasduino-Board-Definitions-for-Arduino-IDE/issues/11#issuecomment-1763796093).

dogerber, 2023-11-04:
I would not expect for

```
  display.clearDisplay();
  display.display();
```

to give anything. What happens if you try this?
```

 display.clearDisplay();
display.println("test");
 display.display();
```

rsoric, 2023-11-07:
Hi @dogerber 

Functions for printing text, drawing shapes and images and clearDisplay - they actually modify the frame buffer which is an array of bytes which are supposed to represent what gets pushed to be shown on the display. display.display() or display.partialUpdate() will actually trigger the sequence of events to send that frame buffer to the display which will trigger the picture changing. Thus, display.clearDisplay followed by display.display should change show a blank screen (all white pixels) on Inkplate. if you were to just call clearDisplay it wouldn't do anything.

Let me know if you're having trouble setting this up, I'd be happy to help
-Rob

Also, I'll be closing this issue as I should have closed it earlier, it's a duplicate.

closed 2023-11-07 by rsoric