SolderedElectronics/Inkplate-Arduino-library #226: display.drawImage() messes up memory in OpenWeather_Station example

dogerber, 2023-11-08:
Hi,

The example supplied in examples/Inkplate10/Projects/Inkplate10_OpenWeather_Station.ino is broken. When run as supplied it shows only one preview image of the weather condition and ignores the rest (the code still runs though).

I was able to pin-point it to this line: 
https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/cbb736fc72e2adddcfdcfa80cec47b7029a8c8ae/examples/Inkplate10/Projects/Inkplate10_OpenWeather_Station/Inkplate10_OpenWeather_Station.ino#L387

when ```display.drawImage(Output, dayPos + 10, dayOffset, true, true); ``` is called, something happens to ```OWOC```, such that all fields containing strings get corrupted. This can be checked by printing ```Serial.println(OWOC.forecast[day].icon)``` before and after the above mentioned line. Before it will be find, afterwards it will be corrupted. 

I do not understand these things enough to fix them and would be happy if someone has an idea.

rsoric, 2023-11-09:
Hi @dogerber 

thanks for bringing this to our attention, we'll take a look as soon as we can. After a short discussion about this with the team, it's likely that the example will be re-written to better reflect our new standards with Inkplate examples. 

I'll update you here when there's progress.

-Rob

sylwester-, 2024-02-07:
Hi,

Any progress on this? just tried it with Inkplate2 and it's not starting up

rsoric, 2024-02-12:
Hi @sylwester- , we unfortunately haven't gotten around to it just yet. Thanks for reminding us, I'll try to squeeze this in our schedule again.

Specifically for Inkplate2, what does the Serial monitor say? Does the device just restart or it can't connect to the internet? There should be some relevant info on the Serial monitor

sylwester-, 2024-02-12:
Looks like this is more on the Netatmo side. I integrated weather station with HomeAssistant and now calling local API and it's been rock solid for 3 days so far without a hiccup.

rsoric, 2024-03-25:
Closing this as there's no more activity and it seems to work.

closed 2024-03-25 by rsoric