SolderedElectronics/Inkplate-Arduino-library #228: Inkplate6PLUS *without* Second I2C GPIO Extender: examples fail

cydergoth, 2023-11-16:
Hi 

I was unlucky enough to get one of the Inkplate6Plus which is missing the second I2C GPIO extender. That in itself is not an huge problem for my project as the project doesn't require it, however none of the examples work on it. I tracked down the first issue to a hang on boot in line 109 of Inkplate6plus.cpp which was causing the startup to fail as it couldn't read the registers from the no-existent GPIO extender. Removing that line (and the ones setting the pin modes + levels) lets the startup complete (also edited to add the USB only power option as per the directions), but the screen doesn't power up and the touchscreen init returns a failure. 

I know the hardware is good because it was working with the factory flashed peripheral mode. 

So two requests, please provide a #define around the init of the second I2C GPIO extender so I don't have to modify the library (makes upgrades harder!) and second what other parameters are likely to be causing the examples to fail?

cydergoth, 2023-11-16:
After some more digging and some I2C scan code injected into the begin(), I can't see I2C device 0x48 (I believe this is the power controller) on the I2C bus after the code which is supposed to ask device 0x20 (the GPIO extender) to toggle the line (GPA3 WAKEUP) which should enable it. 

```
bool Inkplate::begin(void)
{
    if (_beginDone == 1)
        return 0;

    Wire.begin();

#ifndef ARDUINO_INKPLATECOLOR
    for (uint32_t i = 0; i < 256; ++i)
        pinLUT[i] = ((i & B00000011) << 4) | (((i & B00001100) >> 2) << 18) | (((i & B00010000) >> 4) << 23) |
                    (((i & B11100000) >> 5) << 25);
#endif

#ifdef ARDUINO_ESP32_DEV
    digitalWriteInternal(IO_INT_ADDR, ioRegsInt, 9, HIGH);
#else
    digitalWriteInternal(IO_INT_ADDR, ioRegsInt, 9, LOW);
#endif

    memset(ioRegsInt, 0, 22);
    memset(ioRegsEx, 0, 22);

    ioBegin(IO_INT_ADDR, ioRegsInt);
    ioBegin(IO_EXT_ADDR, ioRegsEx);

    pinModeInternal(IO_INT_ADDR, ioRegsInt, VCOM, OUTPUT);
    pinModeInternal(IO_INT_ADDR, ioRegsInt, PWRUP, OUTPUT);
    pinModeInternal(IO_INT_ADDR, ioRegsInt, WAKEUP, OUTPUT);
    pinModeInternal(IO_INT_ADDR, ioRegsInt, GPIO0_ENABLE, OUTPUT);
    digitalWriteInternal(IO_INT_ADDR, ioRegsInt, GPIO0_ENABLE, HIGH);

    WAKEUP_SET;
    delay(1);

   byte error, address;
   int nDevices = 0;

  Serial.println("Scanning for I2C devices ...");
  for(address = 0x01; address < 0x7f; address++){
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0){
      Serial.printf("I2C device found at address 0x%02X\n", address);
      nDevices++;
    } else if(error != 2){
      Serial.printf("Error %d at address 0x%02X\n", error, address);
    }
  }
  if (nDevices == 0){
    Serial.println("No I2C devices found");
  }

    Wire.beginTransmission(0x48);  <-- No device 0x48 on bus at this point
    Wire.write(0x09);
    Wire.write(B00011011); // Power up seq.
    Wire.write(B00000000); // Power up delay (3mS per rail)
    Wire.write(B00011011); // Power down seq.
    Wire.write(B00000000); // Power down delay (6mS per rail)
    Wire.endTransmission();
    delay(1);

    if (pwrMode == INKPLATE_USB_PWR_ONLY)
    {
        // Enable all rails
        Wire.beginTransmission(0x48);
        Wire.write(0x01);
        Wire.write(B00111111);
        Wire.endTransmission();
    }
    else
    {
        WAKEUP_CLEAR;
    }
```

I only see devices on address 0x20 and 0x51, not 0x48

rsoric, 2023-11-20:
Hi @cydergoth ,

which board definition are you using? There is 'e-Radionica' Inkplates and 'Soldered' inkplates included in our board definition package. Please give the other board definition a try (the different one from the one you're using), this sounds like a problem with using the wrong board definition. I'm glad you tested the peripheral mode while it was still flashed to rule out a hardware failure. Let me know how switching the board def works out and we'll see from there.

-Rob

cydergoth, 2023-11-20:
Hey! You're absolutely correct! I switched to the other profile and it
works! Thanks! It was confusing because the factory screen image has the
"Soldered" logo on it, but the e-radionica profile works!

cydergoth, 2023-11-20:
> <SolderedElectronics/Inkplate-Arduino-library/issues/228/1818444524@
> github.com>

Confirmed the touchscreen works too!

Feel free to close this ticket, but it would be ideal if we could have a
define to disable init of the second (missing GPIO extender).

rsoric, 2023-11-21:
@cydergoth really glad to hear it works, and thanks for the suggestion.

closed 2023-11-21 by rsoric