SolderedElectronics/Inkplate-Arduino-library #234: Inkplate10 failing to run Arduino examples

johnholl, 2024-01-11:
Running Arduino IDE 2.0.4
Confirmed that the board is a Soldered Inkplate 10 (not e-radionica) by checking that it has the PCAL expander.
Inkplate library version 8.2.1

When I run Inkplate10_Black_And_White.ino with board set to Soldered Inkplate 10, the display does not change, and the board continually drops and reestablishes the connection on ttyUSB0. I added some Serial.println() statements and found that the crash happens at display.display() in the setup() method. The e-ink display does not change at all.

Even though it has the PCAL expander, I tried also to upload the script with board set to e-radionica.com Inkplate 10 just in case I was misunderstanding the documentation. With some print statements I was able to see that this runs through the entire setup, and begins to loop(). However, the display also never changes in this case.

I've tried all of the above with the grayscale example and several example projects. The behavior is the same in all cases.

rsoric, 2024-01-11:
Hi @johnholl ,

great you already noticed the board version regarding the expander, that's a great first step in solving this and is usually what needs to be changed to fix this. From your description as to what happens, your board is definitely a Soldered Inkplate10 - we just have to figure out why it's crashing.

When looking at the Serial output, is there an ESP32 exception which describes the cause of the crash? There ought to be one, at least saying it's a POWERON_RESET. This is something that could be used to further debug this, could you please copy-paste an example of the Serial output when it crashes, and also use the [ESP exception decoder](https://github.com/me-no-dev/EspExceptionDecoder) to see exactly what's causing the crash in more detail (you can just send me a screenshot of the output you get from this, the tutorial on how to install and use it is in the repository on the link).

That would be a great first step. If you want to test some things ahead of time, I'd for sure keep trying the basic black and white sketch, and the basic grayscale sketches - these sketches should always refresh the display regardless of WiFi or any other factors - so they're great for testing. Upload one of the basic sketches and then maybe try powering Inkplate with a different power source, it's standard 5V USB so any charger can power Inkplate - maybe supplying more power than your computer can is required - this sometimes fixes issues like this. You could also try a different USB cable.

Try these things and let me know

-Rob

johnholl, 2024-01-11:
Hi Rob,

Thanks for getting back to me so quickly. I was able to get the device working when powered directly from a wall adapter so I'm guessing the issue was my laptop's power source not supplying enough/consistent power.

For future reference though, I wasn't seeing any serial monitor output during the "crash". It may not have been crashing, but instead just hanging on display.display().

Since the issue is resolved I'll close it. Thanks again :)

closed 2024-01-11 by johnholl

rsoric, 2024-01-12:
Hi John, great to hear that was it! Hope you enjoy making stuff with Inkplate.