SolderedElectronics/Inkplate-Arduino-library #235: Inkplate6 (e-radionica) 3-bit waveform advice

arduinomnomnom, 2024-01-11:
I have an Inkplate6 (the e-radionica version). With the 3-bit display, the greyscales were unevern and two steps had their intensities swapped, so JPGs looked bad. Changing the VCOM voltage improved the deepness of the blacks, but not the gradiation.

Through blind gueswork, I created modified waveforms (see below) that seem to work ok and give better photo results than the originals.

My questions are:

1. Is there a risk that I might damage the display through a "bad" waveform?
2. How actually do the waveforms work on the display? The sequences of black, white or off do not always make sense, and sometimes a small change in one greyscale step affects the others.

I have Googled for details, but all that I've found is that waveforms are closely guarded secrets and that for the Inkplate displays they had to be reverse-engineered.

```
C:\Users\owner\Documents\Arduino\libraries\InkplateLibrary\src\boards\Inkplate6.h
#ifdef ARDUINO_ESP32_DEV
#define WAVEFORM3BIT \
   { { 0, 1, 1, 0, 1, 1, 1, 0, 0 }, \
     { 0, 1, 0, 1, 0, 1, 1, 0, 0 }, \
     { 0, 1, 0, 1, 1, 2, 1, 0, 0 }, \
     { 1, 1, 1, 2, 2, 1, 0, 0, 0 }, \
     { 0, 0, 1, 2, 1, 1, 2, 0, 0 }, \
     { 2, 2, 1, 1, 2, 1, 2, 0, 0 }, \
     { 1, 0, 1, 2, 1, 2, 2, 0, 0 }, \
     { 0, 0, 0, 0, 0, 0, 2, 0, 0 } };
#endif
```

rsoric, 2024-01-11:
Hi @arduinomnomnom,

I think this is a question for our very own ePaper expert, @BornaBiro can you help?

BornaBiro, 2024-01-16:
Hello @arduinomnomnom!

Yes, damaging the screen is always possible because of using the wrong waveform, so be careful while tweaking it.
Things to note are:
- do not use too many repetitive patterns (usually 3 or 4 of the same color can do a burn-in of the specific color)
- 0 must always be the last phase in the selected color (it's kind of like a discharge, removes charge from the electrodes and makes pixels look sharper)
- 3 is not used (it's for skipping pixel)
- you can always do a more phase, but the refresh time will be slower

About the waveforms, let's first discuss how the waveform even works:

```
          Waveform phases ->
     { 0, 1, 1, 0, 1, 1, 1, 0, 0 }
C    { 0, 1, 0, 1, 0, 1, 1, 0, 0 }
o    { 0, 1, 0, 1, 1, 2, 1, 0, 0 }
l    { 1, 1, 1, 2, 2, 1, 0, 0, 0 }
o    { 0, 0, 1, 2, 1, 1, 2, 0, 0 }
r    { 2, 2, 1, 1, 2, 1, 2, 0, 0 }
s    { 1, 0, 1, 2, 1, 2, 2, 0, 0 }
     { 0, 0, 0, 0, 0, 0, 2, 0, 0 }
```

As you can see, each row represents one color and each column in this color represents one waveform phase. Each phase can have one of the 4 numbers:
0 = Discharge
1 = Make the current pixel darker
2 = Make the current pixel brighter
3 = Skip pixels

So by changing these numbers in the waveform, you can change the appearance of each color. There are more advanced stuff like how to avoid noise on each color and dark/white lines at the edge of the color change but this is more complicated to explain and sometimes not avoidable on Inkplates. Setting the waveform is done by trial and error and comparing it to the reference since as you already said, waveforms are trade secrets and heavily close sourced.

The last thing what you said is that small change can make a big difference in color appearance. The reason behind that is the refresh rate or to be more precise, frame rate. The epaper panel should work at an 85Hz frame rate, so each waveform phase should only last about 12ms. Since the whole framebuffer is stored in PSRAM and there is a lot of conversations from pixels in the framebuffer to the actual data for the epaper panel itself, it takes a lot of bandwidth and processing power that ESP32 just does not have (napkin math; 800 x 600 / 2 = 240000 bytes = ~20MB/s. This is a lot of that that ESP32 and PSRAM need to process). So in reality, we only get 33Hz, but this is plenty enough to get 8 colors. If we could get 85Hz, we could get 4 bit color and much more.

I hope this answers your question to get a better understanding of the waveforms on the Inkplate board.

arduinomnomnom, 2024-01-16:
Hi @BornaBiro,

Thank you for all that info - very useful and I appreciate your time in answering.

Based on your new info, I've made some changes to my waveform, mainly to remove the repetitions of 3. I'm very happy with the result. Next project waiting on my desk: Inkplate10!

```
#ifdef ARDUINO_ESP32_DEV
#define WAVEFORM3BIT \
   { { 1, 0, 1, 1, 0, 1, 1, 0, 0 }, \
     { 0, 1, 0, 1, 0, 1, 1, 0, 0 }, \
     { 0, 1, 0, 1, 1, 2, 1, 0, 0 }, \
     { 1, 1, 2, 1, 2, 1, 0, 0, 0 }, \
     { 0, 0, 1, 2, 1, 1, 2, 0, 0 }, \
     { 2, 2, 1, 1, 2, 1, 2, 0, 0 }, \
     { 1, 0, 1, 2, 1, 2, 2, 0, 0 }, \
     { 2, 0, 0, 0, 0, 2, 2, 0, 0 } }; 
#endif
```

closed 2024-01-16 by arduinomnomnom