SolderedElectronics/Inkplate-Arduino-library #236: Soldered Inkplate 6Color display.display() takes approximately 30 seconds

b-morgan, 2024-01-11:
According to the table at [https://inkplate.readthedocs.io/en/latest/features.html#soldered-inkplate-comparison-newer-models](https://inkplate.readthedocs.io/en/latest/features.html#soldered-inkplate-comparison-newer-models), the Soldered Inkplate 6COLOR refresh time is listed as 12s. Other sources on the internet seem to back this claim.

I am measuring a refresh time of approximately 30 seconds. This is across multiple sketches. I am using the ip6ColorSleepFix branch of the library. I would like to know if you can explain the difference.

I have replaced the `display.display()` function call in multiple sketches with a call to the following function. I have also seen this 30 second timing from comparing timestamped serial monitor output.
``` 
// Function for timing display.display
void timeDisplay()
{
    unsigned long b = millis();
    display.display();
    Serial.print("display.display took ");
    Serial.print((millis() - b) / 1000.0, 3);
    Serial.println(" seconds");
}
```
I have attached a zip file with one of my modified sketches, Inkplate6COLOR_Hourly_Weather_Station.ino and Network.cpp (as I reside in Colorado Springs, Colorado, USA). I have removed my WiFi credentials and my API key.

Also included in the zip file is a picture of my board and the serial output from the running sketch. 

Example output:
```
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2024.01.11 13:26:57 =~=~=~=~=~=~=~=~=~=~=~=
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0

Waiting for WiFi to connect.... connected
Waiting for NTP time sync: ..........
Current time: Thu Jan 11 20:27:04 2024

Fetching data
13d
04d
13d
04d

display.display took 28.472 seconds
ets Jul 29 2019 12:21:46

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0

Waiting for WiFi to connect.... connected
Waiting for NTP time sync: 
Current time: Thu Jan 11 20:28:10 2024

Fetching data
13d
04d
13d
04d

display.display took 28.462 seconds

```
[Temp2.zip](https://github.com/SolderedElectronics/Inkplate-Arduino-library/files/13908262/Temp2.zip)

rsoric, 2024-01-12:
Hi Brad,

thanks for a detailed issue report, we will test this and investigate in more detail and get back to you during next week.

-Rob

b-morgan, 2024-01-15:
Just out of curiosity (and to give you another test point), I installed MicroPython and instrumented the basicCOLOR.py example as follows:
```
@@ -5,6 +5,7 @@
 # Include all the required libraries
 from inkplate6COLOR import Inkplate
 from image import *
+import time
 
 # Create Inkplate object
 display = Inkplate()
@@ -44,4 +45,6 @@
     display.drawBitmap(10, 160, image, 576, 100, display.BLUE)
 
     # Show on the display
+    start = time.ticks_ms()
     display.display()
+    print(time.ticks_diff(time.ticks_ms(), start))
```
Running the modified example produced the output `27825` which is 27.825 seconds consistent with the results previously reported.

rsoric, 2024-02-01:
Hi Brad, 

sorry for the late reply, was busy with some projects here at Soldered.

I've looked into the documentation a bit and due to some panel sourcing issues, some Inkplate 6COLOR models have this slower refresh time, so It's a hardware limitation of that specific ePaper panel and I'm afraid not much can be done about it.

closed 2024-02-01 by rsoric