SolderedElectronics/Inkplate-Arduino-library #237: Upload issue

MiniCodeMonkey, 2024-01-29:
Just got a brand new Soldered Inkplate 10, I'm trying to use it with the Arduino library.

But no matter what I try I end up with an error after the "Configuring flash size..."

Here's the fulll compile/build output:

```
Sketch uses 952665 bytes (30%) of program storage space. Maximum is 3145728 bytes.
Global variables use 55472 bytes (16%) of dynamic memory, leaving 272208 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.2.1
Serial port /dev/cu.usbserial-11520
Connecting.....Traceback (most recent call last):
  File "esptool.py", line 34, in <module>
  File "esptool/__init__.py", line 1004, in _main
  File "esptool/__init__.py", line 790, in main
  File "esptool/loader.py", line 1108, in flash_set_parameters
  File "esptool/loader.py", line 406, in check_command
  File "esptool/loader.py", line 375, in command
  File "esptool/loader.py", line 307, in read
StopIteration
[39160] Failed to execute script 'esptool' due to unhandled exception!

Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:e6:86:a8:2c:cc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Failed uploading: uploading error: exit status 1
```

Here's what I tried:
* Re-install and making sure that I have the newest version of board definitions and library
* Double checking that I selected the correct board and port
* Tried running `inkplate-micropython` which runs fine on the bord
* Tried switching USB ports

Anything I'm missing?

closed 2024-01-30 by MiniCodeMonkey

rsoric, 2024-01-31:
Hi @MiniCodeMonkey 

glad you got it sorted!

gregsadetsky, 2024-03-01:
Thanks a lot @MiniCodeMonkey that solved my issue as well! 115200 was the correct baud rate for me as well (on macos 14.3.1)

Should https://github.com/SolderedElectronics/Inkplate-documentation/blob/master/source/get-started.rst?plain=1#L12 be updated to state that lower speeds should be tried first..? When I saw that line "To tweak your upload speed, you can set Upload speed to 921600.", I thought that it meant that 921600 was the "correct" speed -- my bad obviously.

Happy to open a PR on the documentation if it's helpful.

rsoric, 2024-03-04:
@gregsadetsky just changed it, thanks for the suggestion!