SolderedElectronics/Inkplate-Arduino-library #240: Example google_calender "Guru Mediation Error"

DaMusikant, 2024-02-24:
Hi,
I tested the google calender example for my inkplate6.

But, it always fails with the following code:

`Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008c584  PS      : 0x00060f30  A0      : 0x800d2c95  A1      : 0x3ffb24e0  
A2      : 0x3ffb265c  A3      : 0x00000008  A4      : 0x00000010  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000004  
A10     : 0x3ffb265c  A11     : 0x000000ff  A12     : 0x80092a60  A13     : 0x3ffb2640  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4008c581  LEND    : 0x4008c59b  LCOUNT  : 0xffffffff  

Backtrace:0x4008c581:0x3ffb24e00x400d2c92:0x3ffb24f0 0x400d3253:0x3ffb2780 0x400d3491:0x3ffb27f0 0x400de5ce:0x3ffb2820 `

 What can the cause be? I tried my link in the browser and that works fine.

rsoric, 2024-02-27:
Hi @DaMusikant ,

Could you please use the ESP exception decoder to see what's causing the reset? Here's a link to the ESP exception decoder, it has a tutorial in the README: https://github.com/me-no-dev/EspExceptionDecoder

-Rob

DaMusikant, 2024-03-01:
Hi,

finally I had time to look into it again.

As a side note: EspExceptionDecoder does not work on Arduino Ide >2.0. I had to install 1.8 to get it running.

Anyway, here is the input:

```
Core  1 register dump:
PC      : 0x40166ba6  PS      : 0x00060f30  A0      : 0x8016a069  A1      : 0x3ffb26c0  
A2      : 0xb83fc000  A3      : 0xa041cb38  A4      : 0x3ffb261c  A5      : 0x00000001  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0xb83fc000  A9      : 0x3ffb23e0  
A10     : 0x00000019  A11     : 0x3ffc1bba  A12     : 0x000000a0  A13     : 0x3ffc1bfb  
A14     : 0x0000000f  A15     : 0x00000012  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xa041ccbc  LBEG    : 0x4008c3b5  LEND    : 0x4008c3c9  LCOUNT  : 0xfffffff9  

Backtrace:0x40166ba3:0x3ffb26c00x4016a066:0x3ffb26e0 0x4016a145:0x3ffb2700 0x400d3275:0x3ffb2780 0x400d3485:0x3ffb27f0 0x400ddcce:0x3ffb2820 

ELF file SHA256: 0000000000000000
```
...and the output:

```
PC: 0x40166ba6: cmp(void const*, void const*) at /home/johannes/Arduino/inkplate_calender/Inkplate6_Google_Calendar/Inkplate6_Google_Calendar.ino line 419
EXCVADDR: 0xa041ccbc

Decoding stack results
0x40166ba3: cmp(void const*, void const*) at /home/johannes/Arduino/inkplate_calender/Inkplate6_Google_Calendar/Inkplate6_Google_Calendar.ino line 419
0x4016a066: med3 at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/search/qsort.c line 143
0x4016a145: qsort at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/search/qsort.c line 212
0x400d3275: drawData() at /home/johannes/Arduino/inkplate_calender/Inkplate6_Google_Calendar/Inkplate6_Google_Calendar.ino line 466
0x400d3485: setup() at /home/johannes/Arduino/inkplate_calender/Inkplate6_Google_Calendar/Inkplate6_Google_Calendar.ino line 128
0x400ddcce: loopTask(void*) at /home/johannes/.arduino15/packages/Inkplate_Boards/hardware/esp32/7.1.0/cores/esp32/main.cpp line 42
```

This is absolute too high for me, I hope I did it right. Otherwise let me know :)

rsoric, 2024-03-04:
Thank you, that's the correct info. I suggest you keep the ESP exception decoder as it would be useful for some time in the future.

What it seems to me is that maybe your calendar data is too much, does this also happen when you run it on a new empty calendar which just has a few test events? That should work. Unfortunately the .ICAL  data which we get from Google Calendar can sometimes be a bit too much for Inkplate.

DaMusikant, 2024-03-09:
So, had finally time to get into the problem again.

I created a new calender. The other one was 83.5kB, I didn't thought this could cause any issues.
But the new one with one appointment and 455B was loaded successfully.
Size seems to matter. Even though I thought when we get into the MB area it could go wrong.
Maybe it was also the fact, that this calender was pretty old, first entry 2012, latest 2016.

So, issue fixed. Thank you for your help.

closed 2024-03-09 by DaMusikant

rsoric, 2024-03-19:
Hi everyone, the Google Calendar example will be re-written with a better process for parsing the data and should come as an update to the Inkplate library somewhat soon.