SolderedElectronics/Inkplate-Arduino-library #242: Detect if powered by battery or USB

zBart, 2024-03-05:
I was wondering if there is some way to determine if the device is currently powered by a battery or via USB.
For example: to show a charging indicator, or to change how often I put the device into sleep.

Also, is it possible to wake the device when USB power gets plugged in while the device is in deep sleep?
For example: battery level gets below X, I show a message and put the device into sleep for a day, but when the power gets plugged in I'd like it to wake up instantly (so I can show a charging indicator).

rsoric, 2024-03-19:
Hi @zBart , sorry for the delayed reply. 

Unfortunately the first feature you mentioned isn't included in the design of the devices.

The second feature is possible, there is a jumper which you can short which will bypass the power switch. Check the schematic for your Inkplate or let me know which model you have, I can point to where exactly that jumper is.

rsoric, 2024-03-25:
Closing as there's no more activity

closed 2024-03-25 by rsoric