SolderedElectronics/Inkplate-Arduino-library #251: Can't compile Inkplate 10 Examples

mattncsu, 2024-05-08:
I do not know what changed with Arduino IDE, the board definition, or Inkplate library but I found myself unable to compile a program that worked previously.  Thinking it might be an issue with my installation since I have Arduino IDE versions 1.8.15 and 2.3.2 and dozens of libraries installed, I opened up Windows Sandbox and did a clean install of Arduino 2.3.2, installed the inkplate board 8.1.0 and inkplate library 9.1.0.  I opened the Inkplate10_black_and_white example and hit verify but I get this error:

```
[error.txt](https://github.com/SolderedElectronics/Inkplate-Arduino-library/files/15243178/error.txt)

```

mattncsu, 2024-05-09:
Using 1.8.19,  the inkplate10_black_and_white example compiled but the grayscale example failed. Here is the compile log:
[error18.txt](https://github.com/SolderedElectronics/Inkplate-Arduino-library/files/15257202/error18.txt)

rsoric, 2024-05-09:
Hi @mattncsu 

just had a look at this function, i happen to have the latest version of the library (as I've noted, this should compile without issue on older and newer versions) and this is the function in System.cpp which the compiler is talking about:

```
/**
 * @brief                   Set internal capacitor value.
 *
 * @param bool val          0 or 1 which represents 7pF or 12.5 pF.
 */
void System::rtcSetInternalCapacitor(bool val)
{
    Wire.beginTransmission(I2C_ADDR);
    Wire.write(RTC_CTRL_1);
    Wire.endTransmission();

    uint8_t reg;
    Wire.requestFrom(I2C_ADDR, 1);

    if (Wire.available())
    {
        reg = Wire.read();
    }

    if (val)
    {
        reg |= (1 << 0);
    }
    else
    {
        reg &= ~(1 << 0);
    }

    Wire.beginTransmission(I2C_ADDR);
    Wire.write(RTC_CTRL_1);
    Wire.write(reg);
    Wire.endTransmission();
}
```

The compiler is saying ` error: 'reg' may be used uninitialized in this function`

See how the function has a uint8_t reg declaration so this should always be declared. Please check your System.cpp and add the uint8_t reg to try and fix this. Then let me know if it works better, hopefully there aren't any more errors.

-Rob

mattncsu, 2024-05-11:
I checked my system.cpp and it looks exactly like the one you quoted.

mattncsu, 2024-05-11:
I rolled the board definition back to 7.0.0 and got a different error... no idea what is going on here..ha! 
[error700.txt](https://github.com/SolderedElectronics/Inkplate-Arduino-library/files/15284350/error700.txt)

rsoric, 2024-05-13:
Hi @mattncsu ,

that's really strange, I haven't seen that error unfortunately and the only thing I can recommend is reinstalling Arduino IDE, our libraries and our board definition. We have compile tests for this reason, it should compile fine if installed correctly.

closed 2024-05-14 by rsoric

mattncsu, 2024-05-20:
Still not sure what the issue was or why it wasn't fixed in sandbox mode (a clean VM) but I installed the latest board / library on a different machine and was able to compile my code just fine.

New problem though: my screen no longer refreshes. I tested my code and the Inkplate10_Black_and_white & Inkplate10_grayscale examples. They appear to run fine but the screen doesn't refresh. Do I need to run one of the diagnostic sketches?

rsoric, 2024-05-21:
Hi @mattncsu , which board definitions are you using? The 'Soldered Inkplate' ones or the 'e-Radionica' ones? Only one of these is going to work on your board. If your board has a purple PCB and is newer, it's more than likely the Soldered one.

mattncsu, 2024-05-21:
I used the Soldered definition, the board shipped Sept 2021 via CrowdSuoply, I'll open the case and take a look to tonight.

mattncsu, 2024-05-21:
That fixed it, it was an 'e-Radionica' board. Thanks!

rsoric, 2024-05-22:
Great to hear :) hope you enjoy making cool stuff with Inkplate :)