SolderedElectronics/Inkplate-Arduino-library #255: e-radionica Inkplate 10 link errors

Timmmm, 2024-08-04:
Using the most recent Arduino IDE (2.3.2) I enabled the Inkplate library (10.0.0) and then tried to compile ("verify" 🤢) the Inkplate10_Black_and_White.ino`. With `Soldered Inkplate10` selected as the board it compiles successfully. With `e-radionica.com Inkplate10` I get a load of linker errors. It seems to be basic Arduino stuff like `delay`.

I *really* don't want to have to dig into Arduino's build system. Any idea why this doesn't work? Feels like CI should catch this sort of thing but I can't see how I've done anything wrong.

```
Linking everything together...
"C:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\Inkplate_Boards\\tools\\xtensa-esp32-elf-gcc\\esp-2021r2-patch5-8.4.0/bin/xtensa-esp32-elf-g++" "-Wl,--Map=C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528/Inkplate10_Black_And_White.ino.map" "-LC:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\Inkplate_Boards\\hardware\\esp32\\8.1.0/tools/sdk/esp32/lib" "-LC:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\Inkplate_Boards\\hardware\\esp32\\8.1.0/tools/sdk/esp32/ld" "-LC:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\Inkplate_Boards\\hardware\\esp32\\8.1.0/tools/sdk/esp32/qio_qspi" -T esp32.rom.redefined.ld -T memory.ld -T sections.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.peripherals.ld -mlongcalls -Wno-frame-address -Wl,--cref -Wl,--gc-sections -fno-rtti -fno-lto -Wl,--wrap=esp_log_write -Wl,--wrap=esp_log_writev -Wl,--wrap=log_printf -u ld_include_hli_vectors_bt -u _Z5setupv -u _Z4loopv -u esp_app_desc -u pthread_include_pthread_impl -u pthread_include_pthread_cond_impl -u pthread_include_pthread_local_storage_impl -u pthread_include_pthread_rwlock_impl -u include_esp_phy_override -u ld_include_highint_hdl -u start_app -u start_app_other_cores -u __ubsan_include -Wl,--wrap=longjmp -u __assert_func -u vfs_include_syscalls_impl -Wl,--undefined=uxTopUsedPriority -u app_main -u newlib_include_heap_impl -u newlib_include_syscalls_impl -u newlib_include_pthread_impl -u newlib_include_assert_impl -u __cxa_guard_dummy -DESP32 -DCORE_DEBUG_LEVEL=0 -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-strategy=memw -DARDUINO_USB_CDC_ON_BOOT=0 -Wl,--start-group "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\sketch\\Inkplate10_Black_And_White.ino.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\Inkplate.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\boards\\Inkplate10.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\boards\\Inkplate2.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\boards\\Inkplate4.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\boards\\Inkplate4TEMPERA.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\boards\\Inkplate5.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\boards\\Inkplate5V2.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\boards\\Inkplate6.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\boards\\Inkplate6Color.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\boards\\Inkplate6FLICK.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\boards\\Inkplate6plus.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\boards\\Inkplate7.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\Buzzer.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\Esp.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\Frontlight.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\Graphics.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\Image.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\ImageBMP.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\ImageDither.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\ImageDitherColor.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\ImageJPEG.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\ImagePNG.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\ImageUtils.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\Mcp.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\NetworkClient.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\Pcal.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\Shapes.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\ShapesPolygon.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\System.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\TouchCypress.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\include\\TouchElan.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\libs\\APDS9960\\src\\APDS9960-SOLDERED.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\libs\\APDS9960\\src\\libs\\SparkFun_APDS9960\\src\\SparkFun_APDS9960.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\libs\\Adafruit-GFX-Library\\Adafruit_GFX.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\libs\\Adafruit-GFX-Library\\glcdfont.c.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\libs\\BME680\\src\\BME680-SOLDERED.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\libs\\BME680\\src\\libs\\BME680\\src\\Zanshin_BME680.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\libs\\BQ27441\\src\\BQ27441-G1-SOLDERED.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\libs\\BQ27441\\src\\libs\\SparkFun_BQ27441_Arduino_Library\\src\\SparkFunBQ27441.cpp.o" "C:\\Users\\Tim\\AppData\\Local\\Temp\\arduino\\sketches\\26F1F22D6333D59F0D1AC3B11E4D0528\\libraries\\InkplateLibrary\\libs\\LSM6DS3\\src\\LSM6DS3-SOLDERED.c [... cut ...]
C:\Users\Tim\Documents\Arduino\Inkplate10_Black_And_White/Inkplate10_Black_And_White.ino:586: undefined reference to `Print::print(String const&)'
c:/users/tim/appdata/local/arduino15/packages/inkplate_boards/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Tim\AppData\Local\Temp\arduino\sketches\26F1F22D6333D59F0D1AC3B11E4D0528\sketch\Inkplate10_Black_And_White.ino.cpp.o: in function `loop()':
C:\Users\Tim\Documents\Arduino\Inkplate10_Black_And_White/Inkplate10_Black_And_White.ino:281: undefined reference to `String::String(char const*)'
...
```

Timmmm, 2024-08-04:
Ugh cancel that. It's just some corrupted build issue (maybe I killed it mid-build or something).

Unfortunately due to Arduino being complete shit there's no way to `make clean`, but you can delete `~/AppData/Local/arduino*` and then it works.

closed 2024-08-04 by Timmmm

rsoric, 2024-08-06:
Hi @Timmmm ,

glad you go t it working. Yes, deleting the Arduino15 appdata folder does clean out everything regarding the compiler and fixes issues like this. Arudino has it's quirks, but we still love it :)

-Rob