SolderedElectronics/Inkplate-Arduino-library #256: Google calendar crashes on e-radionica Inkplate 10

Timmmm, 2024-08-04:
Using the e-radionica Inkplate 10, latest Arduino 2.3.2, Inkplate 10.0.0, unmodified Google Calendar example, except to set the wifi details and calendar URL I get this:

```
........
Current time: Sun Aug  4 15:37:08 2024
Getting data... 
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d5112  PS      : 0x00060f30  A0      : 0x800d373e  A1      : 0x3ffb2000  
A2      : 0x0000000a  A3      : 0x00000094  A4      : 0x00000043  A5      : 0x00000008  
A6      : 0x00000043  A7      : 0x3ffd0258  A8      : 0x800d510b  A9      : 0xfffffff4  
A10     : 0x0000000a  A11     : 0x40240000  A12     : 0x00000043  A13     : 0x0000000a  
A14     : 0x7ff00000  A15     : 0x7ff40000  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfffffff4  LBEG    : 0x4008cb29  LEND    : 0x4008cb39  LCOUNT  : 0xffffffff  

Backtrace: 0x400d510f:0x3ffb2000 0x400d373b:0x3ffb2090 0x400d396e:0x3ffb21f0 0x400d3b01:0x3ffb2260 0x400deb7a:0x3ffb2290
```

This isn't going to be fun to debug. :-/ Looks like it is trying to load `0xfffffff4` and that is protected or unmapped. According to [this tool to decode the backtrace](https://github.com/dankeboy36/esp-exception-decoder) which somewhat surprisingly actually worked, it is:

```
PC: 0x400d5112:  is in Shapes::drawThickLine(int, int, int, int, int, float) (c:\Users\Tim\Documents\Arduino\libraries\InkplateLibrary\src\include\Shapes.cpp:190).
EXCVADDR: 0xfffffff4

Decoding stack results
0x400d510f:  is in Shapes::drawThickLine(int, int, int, int, int, float) (c:\Users\Tim\Documents\Arduino\libraries\InkplateLibrary\src\include\Shapes.cpp:190).
0x400d373b:  is in drawEvent(entry*, int, int, int, int*) (D:\workspace\calendar_eink\calendar_eink/calendar_eink.ino:399).
0x400d396e: drawData() at D:\workspace\calendar_eink\calendar_eink\calendar_eink.ino:482
0x400d3b01: setup() at D:\workspace\calendar_eink\calendar_eink\calendar_eink.ino:126
0x400deb7a: loopTask(void*) at C:\Users\Tim\AppData\Local\Arduino15\packages\Inkplate_Boards\hardware\esp32\8.1.0\cores\esp32\main.cpp:42
```

`Shapes.cpp:190` is

```
    fillTriangle(x1, y1, x2, y2, x3, y3, color);
```

Timmmm, 2024-08-04:
```
// Here we store calendar entries
int entriesNum = 0;
entry entries[128];
...
    // Search raw data for events
    while (i < n && strstr(data + i, "BEGIN:VEVENT"))
    {
...
        ++entriesNum;
    }
```

Come on guys. I shouldn't be able to find two school-boy buffer overflows on the very first example I try. This one took me an hour of tedious printf debugging thanks to Arduino's very slow build system.

The hardware is very nice (USB-C!!) but the software is decidedly Arduino-quality. :-/

rsoric, 2024-08-06:
@Timmmm 

Thanks for pointing this out to us, the library is quite big and with all our other projects and products it's hard to keep it up to date with our small team :) We'll be pushing a quick fix for this soon.

-Rob

rsoric, 2024-09-03:
Hi @Timmmm ,

we've recently addressed this in a [push](https://github.com/SolderedElectronics/Inkplate-Arduino-library/commit/f6ef4ce802625c6c6a1afa51fd19b6b47bc4c224) to dev, which will be available in the main library once it's time for a new release. It was resolved by simply stopping to load more than 128 events, thanks to @koderchina who is interning here at Soldered :)

Thanks again for pointing it out to us!

rsoric, 2024-10-11:
That's awesome. We have actually recently made the Soldered Image Converer which you can check out [here](https://github.com/SolderedElectronics/Soldered-Image-Converter). 

The 3-bit image format I have explained in this GitHub [issue](https://github.com/SolderedElectronics/Soldered-Image-Converter/issues/2#issuecomment-2352103490).