SolderedElectronics/Inkplate-Arduino-library #257: Potential buffer overflow in Network.cpp

Timmmm, 2024-08-04:
At least for the Inkplate 10 Google Calendar example your `Network::getData()` function [just loops blindly writing to the output buffer](https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/d33b0e7797eb42fdec34faf164216b547d32cbe3/examples/Inkplate10/Projects/Inkplate10_Google_Calendar/Network.cpp#L122). I can see no bounds checking mechanism.

I thought that might be the cause of #256 but sadly not; my data is like 200k and the buffer is 2MB. Still, you should probably fix that.

rsoric, 2024-08-06:
Hi @Timmmm ,

noted, we will be looking into this and hopefully publish a quick fix in the near future

-Rob

rsoric, 2025-05-16:
Hi @Timmmm , we've updated this example for Inkplate in the latest release, check it out [here](https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/master/examples/Inkplate10/Projects/Inkplate10_Google_Calendar/Inkplate10_Google_Calendar.ino).

closed 2025-05-16 by rsoric