SolderedElectronics/Inkplate-Arduino-library #258: Feature request: Auto-detect image format

jallbrit, 2024-08-15:
First off, thank you for this library, I have had a lot of fun with my Inkplate 6COLOR. It's very cool and this library makes it really easy to work with.

In the `inkplate.drawImage()` [docs](https://inkplate.readthedocs.io/en/stable/arduino.html#inkplate-drawimage), it says:

> Can draw all kinds of images, but they should have a file extensions in them.

But, I'm serving images at an API URL that directly returns an image, so I can't edit the extension... there is no file extension for the API endpoint that my inkplate is reaching out to. And I'm returning random images, so I don't know in advance what type of image it will be (I have a mixture of JPG and PNG).

It would be really great if `inkplate.drawImage()` could auto-detect whether a given image file is PNG, JPG, or BMP *without* needing a file extension. I know the different images types have certain headers that could be read.

jallbrit, 2024-08-15:
Well, I ended up solving this issue by just fixing my API endpoint to convert to PNG before returning an image, so Inkplate can always expect a PNG. But, I'll leave this issue open for the maintainers to decide what to do.

rsoric, 2024-08-19:
Hi @jallbrit,

sorry for a bit a delayed reply on this, I was on PTO.

Glad you're overall enjoying the library :) 

drawImage has an overloaded definition which is able to do this kind of format detection: https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/d33b0e7797eb42fdec34faf164216b547d32cbe3/src/include/Image.cpp#L74C13-L74C22 - and it should work as long as your URL ends in .jpg, .jpeg or .png. Could you try that? 

Or do you mean you just have a raw file and we have to look into the actual bytes to see what format it is?

-Rob

rsoric, 2024-10-22:
Closing due to inactivity, feel free to write back if the issue persists

closed 2024-10-22 by rsoric