SolderedElectronics/Inkplate-Arduino-library #259: RTC not saving time

ajoliveau, 2024-09-08:
I have a problem with my e-radionica Inkplate 6. Whenever I set the RTC, if I try to get rtc data it returns nothing or invalid data.

The [SimpleRTC](https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/master/examples/Inkplate6/Advanced/RTC/Inkplate6_RTC_Simple/Inkplate6_RTC_Simple.ino) example displays `00:00:00 Sunday, 00/00/69`

Running this simple program : 

```
#include "Inkplate.h"            // Include Inkplate library to the sketch
Inkplate display(INKPLATE_1BIT); // Create an object on Inkplate library and also set library into 1-bit mode (BW)

// Set clock
uint8_t hour = 8;
uint8_t minutes = 25;
uint8_t seconds = 0;

// Set date and weekday (NOTE: In weekdays 0 means Sunday, 1 means Monday, ...)
uint8_t weekday = 4;
uint8_t day = 16;
uint8_t month = 12;
uint8_t year = 21;

void setup()
{
    display.begin();        // Init Inkplate library (you should call this function ONLY ONCE)
    display.clearDisplay(); // Clear frame buffer of display
    // display.display();      // Put clear image on display
    display.setTextSize(4); // Set text to be 4 times bigger than classic 5x7 px text

    display.rtcSetTime(hour, minutes, seconds);    // Send time to RTC
    display.rtcSetDate(weekday, day, month, year); // Send date to RTC

    display.println(display.rtcGetEpoch());

    display.display(); // Put frame buffer on display
}
void loop()
{
}
```
always returns 1925740689 which is a date far in the future.
I have a battery connected , but it's the same thing with the battery removed.

I actually had the exact same problem with another unit, I did an RMA for another reason thinking I didn't understand how that worked, so I doubt it's a hardware issue ?

How can I debug this ?

Thank you !

rsoric, 2024-09-12:
Hi @ajoliveau 

sorry for a bit of a delayed reply, going to give a quick reply now but I'll test this further during next week if it's required.

Firstly, could you please try the 'dev' branch of the repository as the installed library? 

And could you please try the entire Inkplate6 rtc simple sketch, just the way it is. Because that has a call to rtcGetData which I think might affect this. If in your sketch, try and call rtcGetData before rtcGetEpoch.

Hopefully that's a quick fix :) If not, we'll have a look at this.

-Rob

rsoric, 2024-09-16:
Hi @ajoliveau 

After consulting with my team, it has been pointed out that the initial e-Radionica Inkplate6 models with touchpads may not even have a RTC. Could you please send a photo of the back of your board?

-Rob

ajoliveau, 2024-09-17:
This is the board
[image]

Looking at the Crowdsupply page, it looks like they don't advertize a RTC :( 
So it's completely a user problem ! Sorry for wasting your time

rsoric, 2024-09-17:
Indeed this is the case. Happy we got to the bottom of it :) 

Hope you'll have fun using Inkplate!

closed 2024-09-17 by rsoric