SolderedElectronics/Inkplate-Arduino-library #261: Inkplate6 (Radionica ) compile error fatal error: bits/c++config.h: No such file or directory

rhousedorf, 2024-10-31:
Hi Folks,

FYI, I have a portable Arduino IDE install set for this board, but it is failing.
**I have not compiled this in at least two years.**
OpenWeather just deprecated their API, thus a recompile is needed.  gotta change the URL/URI whatever..

So, I get a missing file for 
y:\projects\arduino alt hw instances\ardide-ink602_2024\portable\packages\inkplate_boards\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\cmath:41:10: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
 
 ( FYI, this is a COPY of the one that worked last time I built it... )

I read something in another repo about the build output dir being too long for Windows, not sure if you folks are building like that?
Makes it not create those .h file(s) ?

I'm a .NET guy, don't use c++, I have downloaded this repo.
Can I simply build it myself?  will that generate these .h files?
I tried creating an empty c++config.h, but it just asked for more, other, .h files.  I stopped after the 3rd try.....

If I can build it myself, how do I do that?

Appreciate any info you can give.
I've had this inkplate next to my desk since I bought it, and KEEP LOOKING AT IT, all day, every day.
"Network error, probably wrong API key"
is killing my brain.....

BornaBiro, 2024-11-04:
Hi @rhousedorf 
Error about the missing file is related to the Inkplate Board Definition. Can you tell me what version of the Inkplate Board Definition are you using? We had this issue on one of the older versions of the board def.

rsoric, 2024-11-19:
Hi @rhousedorf , any updates? I'll be taking care of the ticket from now on

-Rob

rsoric, 2024-12-06:
Closing due to inactivity

closed 2024-12-06 by rsoric