SolderedElectronics/Inkplate-Arduino-library #262: Network error, probably wrong api key

flyin-wheels, 2025-01-11:
'Network error, probably wrong api key' error displayed on inkplate6 after loading Inkplate6_Hourly_Weather_Station. When I submit URL, lon, lat, api key in a browser I get {
    "cod": 401,
    "message": "Invalid API key. Please see https://openweathermap.org/faq#error401 for more info."
}

I have Inkplate6 and my api works for other examples. Is a subscription required for an hourly update on openweathermap?

flyin-wheels, 2025-01-11:
Am I required to get a OneCall subscription for the new API call using Onecall? https://api.openweathermap.org/data/2.5/onecall?lat=%s&lon=%s&appid=%s

My free subscription and api key were created some years ago. This URL works in a browser - https://api.openweathermap.org/data/2.5/weather?lat=47.60&lon=-122.335167&appid=2d3dd2e4d264e1f87a3226a24640d578.

rsoric, 2025-01-14:
Hi @flyin-wheels,

Yes you should be able to use the free api key if it was made that long ago. That URL (the second one you sent) with your API key does not work for me, I get:

`{"cod":401, "message": "Invalid API key. Please see https://openweathermap.org/faq#error401 for more info."}`

So, check the API key please, if it's correct, unsure why it would work on your network and not on ours if that's the case.

-Rob

flyin-wheels, 2025-01-14:
Thank you for the email. I ended up deleting that account, sorry. It did return a json message in browser.

On the website it seems they want you to use a credit card for a one call account. I just don’t like doing that even if it is free.

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

rsoric, 2025-01-14:
You're welcome @flyin-wheels  

yeah understandable, there are alternative API's, and you can modify the network.cpp file in the example as to how you're getting the data and use the rest of the sketch how it is (the data drawing and deepsleep features mainly).

We are looking into updating the weather examples, we're just quite busy with other projects right now.

As a side recommendation, I've heard great things about [open-meteo](https://open-meteo.com/) which is free and doesn't require the card.

Hope that helps somewhat, as this is an external API issue, there isn't much to fix other than modify the example to use a different API so I'll close this for now.
-Rob

closed 2025-01-14 by rsoric

flyin-wheels, 2025-01-14:
Robert,

Thanks again. No worries, I had the Inkplate in a box for years. I am just now getting it setup.

I will look into other APIs and see what I find.

Jim

Sent from [Proton Mail](https://proton.me/mail/home) for iOS