SolderedElectronics/Inkplate-Arduino-library #263: Version 10.1.0 appears to contain old ESP32 NetworkClientSecure library which breaks included examples.

crossfireprod, 2025-01-15:
Hello! I just unboxed my INKPLATE 6 PLUS and am attempting to try some of the included example projects.  Unfortunately, I can't get any of them to compile due to the following error:

```
/home/zach/Arduino/libraries/InkplateLibrary/src/include/NetworkClient.cpp: In member function 'uint8_t* NetworkClient::downloadFileHTTPS(const char*, int32_t*)':
/home/zach/Arduino/libraries/InkplateLibrary/src/include/NetworkClient.cpp:400:13: error: 'class WiFiClientSecure' has no member named 'setInsecure'
     client->setInsecure(); // Use HTTPS but don't check cert
             ^
```

If I search for instances of `NetworkClient.cpp`, I come up with the following:
```
/home/zach/.arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1/libraries/WiFiClientSecure/src/WiFiClientSecure.h
/home/zach/.arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1/Inkplate_core-1.0.1/libraries/WiFiClientSecure/src/WiFiClientSecure.h
/home/zach/.arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1/Inkplate_core-1.0.1/Inkplate_core-1.0.1/libraries/WiFiClientSecure/src/WiFiClientSecure.h
/home/zach/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/NetworkClientSecure/src/WiFiClientSecure.h
```

The first three hits (all included as a part of the Inkplate library) do not contain the referenced `seTInsecure` function.  The ESP32 library version `3.0.7`, however, does.

Apologies if I'm missing something silly, but it seems there's some sort of oversight in the version of what's included in / required by the Inkplate library?

Many thanks in advance for any insight you can provide.

rsoric, 2025-01-15:
Hi @crossfireprod ,

thanks for pointing this out to us. It seems to me the Inkplate boards package on your machine is called 'Croduino_Boards' and this is an old, outdated version. Could you please verify that you're using 'Dasduino boards' instead, and then try it out.

[Here's a tutorial on how to install them.](https://github.com/SolderedElectronics/Dasduino-Board-Definitions-for-Arduino-IDE?tab=readme-ov-file#dasduino-board-definitions-for-arduino-ide) Also make sure that you have the latest version of the Inkplate library. I believe this should work as we have compile tests set up on this repository which will perform compile tests on every new version of the library and they have passed for the last official release. 

Let me know!

-Rob

crossfireprod, 2025-01-15:
Oh my, not only is Croduino_Boards old and outdated, the provided [README](https://github.com/SolderedElectronics/Croduino-Board-Definitions-for-Arduino-IDE) makes that explicitly clear!  So sorry to have wasted your time.  As expected, the examples compile when the correct library is installed...

closed 2025-01-15 by crossfireprod